freifunk-berlin/firmware #684: policy routing problem when using multiple networks on the same device

pmelange, 2019-05-17:
Hedy 1.0.2 tunneldigger version.

Having a management network running parallel and on the same VLAN as another network results in not being able to route to the second network space.  For example, quitte-core has a normal 10.36.x.x/24 client network.  Additionally there are many AP's on the network, and all have been given an ip in the 192.168.0.0/24 range.

Previously, the firmware was able to route this properly.  Here is what the rule looks like now.
```
root@quitte-core:~# ip r g 192.168.0.21
192.168.0.21 dev br-dhcp  table olsr  src 10.36.x.1 
    cache 
```

It should look like 
```
root@quitte-core:~# ip r g 192.168.0.21
192.168.0.21 dev br-dhcp  src 192.168.0.1 
    cache 
```

I have confirmed with tcpdump that outgoing ping requests are being sent to 192.168.0.21 with a source address of 10.36.x.1.

I have tried setting this up in multiple ways.  I have tried aliases ```option ifname '@dhcp'```, explicit device names ```option ifname 'eth0.10'````, setting up a bridge, and having 2 IP's in the dhcp interface config ```list ipaddr 'xxxx'```.

The workaround is to remove any interfaces which share a device (the second network) from the configuration.  Then, manually add the IP address after boot.
```
ip addr add 192.168.0.1/24 dev br-dhcp
```

The previous firmware image was a hand made ipsec image created by @SvenRoederer with a version number of 1.0.0 alpha.  Having multiple address spaces on one device worked fine.  I believe that since then, the policy routing has been changed quite a bit.

SvenRoederer, 2019-05-19:
* as per https://github.com/freifunk-berlin/firmware/commits/Hedy-1.0.x/patches I don't see any changes to our policyrouting.patch
* this kind of setup seems to be an "individual special configuration" case
* was your old setup using policyrouting at all?

This issue might have to be reported against the external-repo we use.

pmelange, 2019-05-20:
* Since @SvenRoederer configured the policy routing, he should be able to state exactly how the configuration was customized
* Policy routing was enabled in the old setup.
---
the origional ip rule 
```
config rule
	option lookup 'main'
	option priority '200'
	option in 'dhcp'
```
doesn't work anymore.  But this rule does (unfortunately with hardcoded ip range)
```
config rule                                     
        option lookup 'main'        
        option priority '200'                        
        option dest '192.168.0.0/24'             
```

Now that a solution has been found, closing....

closed 2019-05-20 by pmelange