freifunk-berlin/firmware #688: UBNT-ERX: initrd-images don't have address 192.168.42.1

SvenRoederer, 2019-05-31:
```
your code goes here
```

### Bug report

**Booting a initrd-image (freifunk-berlin-dev-daily-5121287-ramips-mt7621-ubnt-erx-initramfs-kernel.bin) has IP-Address 192.168.1.1**
These images should also have the default freifunk-address (192.168.42.1) they listen to.

SvenRoederer, 2019-05-31:
These images are missing the "freifunk-berlin-defaults"-packages

SvenRoederer, 2019-06-02:
Adding the "freifunk-berlin-network-defaults" will pull dependencies of "lib-guard" and "iwinfo". Esp. "iwinfo" pulls again some other packages. All this is not needed in the initrd-images and might cause that they become to large.
Adding these packages will also make the settings being included into the sysbackup created during sysupgrade, which seems not ideal.

So one solution might be to create a separate "freifunk-berlin-initrd-defaults" package which is taking care of the correct settings.

Any comments on this idea?

pmelange, 2019-06-03:
Is it not possible to set the IP with 

```
CONFIG_TARGET_PREINIT_IP="192.168.42.1"
CONFIG_TARGET_PREINIT_NETMASK="255.255.255.0"
CONFIG_TARGET_PREINIT_BROADCAST="192.168.42.255"
```

SvenRoederer, 2019-06-03:
As mentioned in https://github.com/freifunk-berlin/firmware/commit/398aee068422f0ae88964e776dc9df39883d1aa6 the PREINIT_* settings only apply to preinit & failsafe

SvenRoederer, 2019-06-17:
This also affects the "omnia-medkit" recovery of the Turris Omnia

pmelange, 2019-06-18:
The ultimate solution might be to have the IP address stored in /etc/board.json.
```
	"network": {
		"lan": {
			"ifname": "eth0",
			"protocol": "static",
			"ipaddr": "192.168.42.1" # this line is normally not there
		}
	}
```

We would have to add "ipaddr".  This in turn will be used by https://github.com/openwrt/openwrt/blob/aa3f9736ea67200dd840093b848606ced27d388e/package/base-files/files/bin/config_generate#L99.  If "ipaddr" is not set, then 192.168.1.1 is used.

The only thing I haven't figured out yet is how to get this into board.json.  Any ideas?

SvenRoederer, 2019-06-18:
as far as I remember board.json will be generated by some script during initial-bootup.
But it seems much more portable to create a separate package or split up current network-defaults package, than patching inside OpenWrt to solve this.

SvenRoederer, 2020-03-26:
It might help, just to add the "freifunk-berlin-network-defaults" package to the initrds, but this will also pull many additional packages as dependency.