freifunk-berlin/firmware #689: UBNT-ERX: missing factory.tar images

SvenRoederer, 2019-06-01:
### Bug report

**there are no factory.tar images to flash the board from out of the vendor OS**

Currently we generating initrd-boot images for these routers ```*-ramips-mt7621-ubnt-erx-initramfs-kernel.bin``` for TFTP-boot. OpenWrt seems to be able to generate ```factory.tar``` files, see https://github.com/openwrt/openwrt/blob/a1210f88882415f629cec272a0017fcb5a2ced3c/target/linux/ramips/image/mt7621.mk#L457 .

Providing these images will not force the user to open their device and use a serial-adapter to operate the boot-loader.

SvenRoederer, 2019-06-04:
With the following OpenWrt config  [erx-factory-initrd-tar.txt](https://github.com/freifunk-berlin/firmware/files/3249696/erx-factory-initrd-tar.txt) I was able to build a factory.tar, based on non-recent OpenWrt-master. As I don't had a device with vendor firmware avail, I wasn't able to test functionality.

pmelange, 2019-06-04:
To convert back to the factory, Ubiquiti has released this recently

https://help.ubnt.com/hc/en-us/articles/360018189493-EdgeRouter-Manual-TFTP-Recovery

it works great and even rescued an ERX-SFP with a bad block on the flash.

SvenRoederer, 2020-10-20:
digging around brings https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=201dbb9fe8f61572d66f02942c4482af9b4f2f0c which adds a test that the "ubnt-erx-factory-image" will not be larger than "KERNEL_SIZE" (3MB).
Our current initrd, based on OpenWrt-master (linux 5.4), is around 3342654 bytes (2MB kernel, 1,2MB root-fs) which is above this limit. The list of packages installed in the image is already very small ([ubnt-erx-sfp.manifest](https://github.com/freifunk-berlin/firmware/files/5411990/ubnt-erx-sfp.manifest.txt)) so reducing it even more will become hard. I see 2 options: 
* reducing the size of the kernel - If we don't want to use a separate kernel-config for the factory-images, which in turn requires rebuilding OpenWrt 2 times.
* avoid using LuCI and luci-mod-failsafe - This might reduce the image enough (no lua, luci, rpcd and relating libs), but requires to write some custom CGI for a web-based sysupgrade.

To solve this issue, the 2nd option seems the most effective.

SvenRoederer, 2020-10-20:
> rescued an ERX-SFP with a bad block on the flash.

@pmelange I'm wondering, how reflashing will fix bad flash-cells ...

SvenRoederer, 2020-10-22:
> * reducing the size of the kernel - If we don't want to use a separate kernel-config for the factory-images, which in turn requires rebuilding OpenWrt 2 times.
>
>     * avoid using LuCI and [...]

AS the ER-X is not the "working Horse" for Freifunk and also "non-Wifi" its not the boards for the average user. The "advanced" user should be confident in using the "OpenWrt-snapshot factory.tar" (with out Web-GUI) or the TFTP (via TTL-RS232-console).

closed 2020-10-22 by SvenRoederer