freifunk-berlin/firmware #690: upgrading a remote node will not make it connecting to the BBB again (tunneldigger)

SvenRoederer, 2019-06-01:
### Bug report

**An upgraded node is missing the tunneldigger package**

When upgrading a remote node via BBB-VPN was working painless with pre-Hedy releases. As the required OpenVPN packages have been included in all images a remote updated node was reconnecting to BBB-VPN by itself.
Since we switched to modular uplinks and an optional alternative BBB-VPN via tunneldigger the old situation can only be archived when using a matching "tunneled uplink" package. When using a no-tunneled image the BBB-VPN link can only be restored after a site-visit an installing tunneldigger / OpenVPN manually.

It would be great to have a working BBB-VPN connection after an update.

pmelange, 2019-06-03:
This issue has been discussed at the weekly cbase meeting and various proposals have been discussed.  Additionally, if the node uses the openvpn variety (luckily there are only about 12 users of this old service left) to connect to the bbb-vpn and notunnel or tunneldigger image is used, the same problem arises.

SvenRoederer, 2019-06-03:
In fact that's something which that closely relates to the flavors of images prebuilt. Probably the attendedsysupgrades, mentioned in https://github.com/freifunk-berlin/firmware/issues/22#issuecomment-498350409, are a smart solution.

pmelange, 2019-06-14:
This has been implemented in https://github.com/freifunk-berlin/firmware/pull/701 but is also dependant on https://github.com/freifunk-berlin/firmware-packages/pull/189