freifunk-berlin/firmware #694: Support more UMTS/LTE sticks

pmelange, 2019-06-04:
I recently received the following email...

ich folge gerade deiner Installationsanleitung zum Betrieb eines Huawei
LTE-Sticks am AR300M (ich habe nicht die Lite-Version). Freifunk ist
geflasht und geht, der Stick geht auch, allerdings habe ich bei der
Installation der zusätzlichen Pakete ein Problem. Etwa die Hälfte der
Pakete ist bei meiner Version von Hedy nicht verfügbar.

Version: hedy-1.0.3-gl-ar300m-sysupgrade.bin (Tunneldigger)

Hier die Konsole:

---

root@resist:~# opkg install kmod-usb-net-rndis comgt-ncm
kmod-usb-net-huawei-cdc-ncm kmod-usb-serial \
> kmod-usb-serial-option kmod-usb-serial-wwan usb-modeswitch kmod-mii
kmod-usb-net \
> kmod-usb-wdm kmod-usb-net-qmi-wwan uqmi kmod-usb-net-cdc-mbim umbim
Package kmod-usb-net-rndis (4.4.167-1) installed in root is up to date.
Unknown package 'comgt-ncm'.
Unknown package 'kmod-usb-net-huawei-cdc-ncm'.
Package kmod-usb-serial (4.4.167-1) installed in root is up to date.
Package kmod-usb-serial-option (4.4.167-1) installed in root is up to date.
Package kmod-usb-serial-wwan (4.4.167-1) installed in root is up to date.
Unknown package 'usb-modeswitch'.
Package kmod-mii (4.4.167-1) installed in root is up to date.
Package kmod-usb-net (4.4.167-1) installed in root is up to date.
Unknown package 'kmod-usb-wdm'.
Unknown package 'kmod-usb-net-qmi-wwan'.
Unknown package 'uqmi'.
Unknown package 'kmod-usb-net-cdc-mbim'.
Unknown package 'umbim'.
Collected errors:
* opkg_install_cmd: Cannot install package comgt-ncm.
* opkg_install_cmd: Cannot install package kmod-usb-net-huawei-cdc-ncm.
* opkg_install_cmd: Cannot install package usb-modeswitch.
* opkg_install_cmd: Cannot install package kmod-usb-wdm.
* opkg_install_cmd: Cannot install package kmod-usb-net-qmi-wwan.
* opkg_install_cmd: Cannot install package uqmi.
* opkg_install_cmd: Cannot install package kmod-usb-net-cdc-mbim.
* opkg_install_cmd: Cannot install package umbim.

---

It would be great if we could support more LTE devices.

SvenRoederer, 2019-06-05:
That LTE-discussion is close to a bordercase. I remember an old discussion about adding a big list of USB-WiFi-sticks, which was rejected as of wasting build-power and diskspace.

booo, 2019-06-06:
It's an edge case, yes.

@pmelange Could you check the size of the kernel modules? Maybe we can just build the kernel modules because they can't be used from a different source. The userland programs can be installed from openwrt.

pmelange, 2019-06-10:
From https://wiki.freifunk.net/Berlin:Freifunk-Router_mit_LTE/UMTS-Uplink the following packages will be installed.  I'm not sure which of the following kmods we are building regardless of support of UMTS/LTE support.

Values are in bytes.  Based on the master branch TARGET=mpc85xx-generic (based on OpenWrt 18.x).  Things might change in 19.x, but probably not by much.

kmod-usb-net-rndis 5247
kmod-usb-serial 12983
kmod-usb-serial-option 5614
kmod-usb-serial-wwan 4495
kmod-mii 3489
kmod-usb-net 11877
kmod-usb-net-huawei-cdc-ncm 2416
kmod-usb-wdm 6422
kmod-usb-net-qmi-wwan 6440
kmod-usb-net-cdc-mbim 4519

Just for info, the userspace programs:
umbim 14936
comgt-ncm 3244
usb-modeswitch 13846
uqmi 29086
luci-proto-3g 2444

SvenRoederer, 2019-10-24:
can we close this as we have merged 21ea0920f7f25d06fbad90c3a7580f9127951186 and e244295f5fd6b3d98741e655aa6b00793dc699a0 ?

SvenRoederer, 2019-12-09:
I assume it's solved, so closing.
If still unsolved, please reopen.

closed 2019-12-09 by SvenRoederer