freifunk-berlin/firmware #695: Archer D50

niccokunzmann, 2019-06-07:
Ich habe 4 Archer D50 bekommen.
Es gibt [2 Commits](https://github.com/openwrt/openwrt/pull/2031/commits) im [master Branch von openwrt](https://github.com/openwrt/openwrt), die die Kompatibilität herstellen.

## Versuch 1
Ich habe versucht, die Commits auf den Commit [c476954633887d8a1fcd80482e821074d7c4a36e](https://github.com/openwrt/openwrt/commit/c476954633887d8a1fcd80482e821074d7c4a36e) aus der [config.mk](https://github.com/freifunk-berlin/firmware/blob/v1.0.3/config.mk#L6)-Datei drauf zu cherry-picken. Es stellt such heraus, dass das ath79-Target dort garnicht existiert.

## Fragen
- Beruht diese Firmware beruht auf dem LEDE-branch?
- Gibt es einen Release, der auf dem OpenWRT-Master aufbaut?
- Wie könnte ich vorgehen, um den Router zu unterstützen?

pmelange, 2019-06-07:
Tags lig v1.0.x are based on 17.x (LEDE)
master branch is based on OpenWrt's openwrt-18.06 branch
daily/upstream-master is based on the Openwrt master branch.

You probably want to use daily/upstream-master

And since these changes have been merged already, you will probably just have to modify the file https://github.com/freifunk-berlin/firmware/blob/daily/upstream-master/profiles/ath79-generic.profiles and add the new target ```tplink_archer-d50-v1``` to the list.

niccokunzmann, 2019-06-13:
@pmelange Thanks! This is something new I learned. It compiled but I could not generate a file for upload.

closed 2019-06-13 by niccokunzmann

pmelange, 2019-06-13:
Sounds great.  Maybe you would like to make a PR to add this device to ath79-generic.profiles?  And which file did you want to generate to upload?

SvenRoederer, 2019-06-13:
Now there is also the branch "daily/upstream-1907" which is based on OpenWrt-19.07. There should also easy integration possible.

niccokunzmann, 2019-06-13:
Currently, there is only a sysupgrade file. There is no file which can
be uploaded via the UI/website of the device.
So, I do not see a use in adding this as not many people will open the
device and attach a serial cable to the pins.
If we would like to have the file to upload through the UI, how would we
go about it?

pmelange, 2019-06-13:
This is the only upstream installation method I have found so far
https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=f5d2c91415a68f554815860d574145644fc31c16
Have you seen anything else?

When there is an upstream factory image, then it will be very easy to make a freifunk factory image.

niccokunzmann, 2019-06-14:
I have not found any other method. So, I would like to wait until there
is a factory image or work towards it.

pmelange, 2019-06-14:
I think it is still worth it to have this device supported.  In the firmware wiki page (https://wiki.freifunk.net/Berlin:Firmware), the entry for this device would only need a fitting footnote.  Perhaps not everyone is ready to open up the router and do it "the hard way", but some might.

Again, I would find it really great if there was a PR submitted to add support for this device.

niccokunzmann, 2019-06-15:
What keeps you from submitting this pull request? What do you need to do
it yourself?

pmelange, 2019-06-15:
Normally I would build it locally and test it on my own device before submitting a pr. But if you want me to, I can at least make a branch with the device support so that you or anyone else with the device can test it.

Additionally I am always happy when people get involved in the firmware development.

I can do the work tomorrow and I will post a link to the firmware here

niccokunzmann, 2019-06-16:
> But if you want me to, I can at least make a branch with the device support so that you or anyone else with the device can test it.

I do not want you to do anything.
Now, it seems, I will do a pull request as soon as I have my devices flashed. I already built the firmware's sysupgrade. I wonder how to create the factory image. That is in progress at openwrt, I guess but do not know.

SvenRoederer, 2019-06-16:
> Now, it seems, I will do a pull request as soon as I have my devices flashed. I already built the firmware's sysupgrade. I wonder how to create the factory image. That is in progress at openwrt, I guess but do not know.

Not sure if @DavideFioravanti or anyone else is trying to generate factory-images at all. To generate a factory-image you need to figure out the internal format and probably deal with signing.

DavideFioravanti, 2019-08-10:
Hello there.
Sorry for my late reply. As far as I know, it's not currently possible to create a factory image because of the V3 header.
You can read more here: https://github.com/xdarklight/mktplinkfw3#missing-parts