freifunk-berlin/firmware #709: gonzo-1.1.0-rc1 still have adhoc as default

strassburger, 2019-06-17:
### Bug report

**What is the problem?**

**What is the expected behaviour?**

**Firmware Version:**

**Site Configuration:**

**Custom patches:**

strassburger, 2019-06-17:
gonzo-1.1.0-rc1 still have adhoc as default
default should be 802.11s
test with:

GL AR-150
freifunk-berlin-development-62ca344-gl-ar150-sysupgrade.bin

strassburger, 2019-06-17:
> gonzo-1.1.0-rc1 still have adhoc as default
> default should be 802.11s
> test with:
> 
> GL AR-150
> freifunk-berlin-development-62ca344-gl-ar150-sysupgrade.bin

pmelange, 2019-06-17:
You used the wrong image. It should be c8948db6c2869d503fcd03c4fa5057c6386d0fd5.

Click on 'Auch alte Releases und Development-Branches anzeigen' to find it.

For your device the link is http://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx-generic/1448/tunnel-berlin-tunneldigger/freifunk-berlin-gonzo-1.1.0-rc1-c8948db-gl-ar150-sysupgrade.bin for the tunneldigger version

strassburger, 2019-06-17:
Thanks, that works fine.
It looks like the link i got from here:
https://lists.berlin.freifunk.net/pipermail/berlin/2019-June/039457.html
is wrong.
the wrong link is this one:
https://buildbot.berlin.freifunk.net/buildbot/unstable/

closed 2019-06-17 by strassburger