freifunk-berlin/firmware #710: gonzo-1.1.0-rc1 freifunk-berlin-bbbdigger not working after update from Hedy 1.0.3

strassburger, 2019-06-17:
Installing freifunk-berlin-bbbdigger on a new installed Gonzo 1.0.0 RC: 
no problem after waiting a while.

After installing freifunk-berlin-bbbdigger on an updated machine:
No access to olsr-devices after updating
On workstation:
ping 10.230.18.1  --> OK
ping chris-core.olsr: Der Name oder Dienst nicht bekannt.

on freifunk-router:
ping 10.230.18.1  --> OK
ping chris-core.olsr:  bad address 'chris-core.olsr'

In Luci:
OLSR | Nachbarn:
OLSR Daemon
Es konnte keine Verbindung zu OLSR-Daemon hergestellt werden

same experience on GI-AR150 with firmware:
freifunk-berlin-gonzo-1.1.0-rc1-c8948db-gl-ar150-sysupgrade.bin

and TL WDR 3600 v1 with firmware:
freifunk-berlin-gonzo-1.1.0-rc1-c8948db-tl-wdr3600-v1-sysupgrade.bin

pmelange, 2019-06-17:
> Installing freifunk-berlin-bbbdigger on a new installed Gonzo 1.0.0 RC:
> no problem after waiting a while.
>
> After installing freifunk-berlin-bbbdigger on an updated machine:
> No access to olsr-devices [...]

How long did you wait?  The hostnames of nodes reachable by olsrd are updated every 5 minutes.  Do you still have this problem?  Take a look in /tmp/hosts/olsr to see if there are any entries.

> 
> on freifunk-router:
> ping 10.230.18.1 --> OK
> ping chris-core.olsr: bad address 'chris-core.olsr'

same as above.

> 
> In Luci:
> OLSR | Nachbarn:
> OLSR Daemon
> Es konnte keine Verbindung zu OLSR-Daemon hergestellt werden
> 
> same experience on GI-AR150 with firmware:
> freifunk-berlin-gonzo-1.1.0-rc1-c8948db-gl-ar150-sysupgrade.bin

What was the version installed before?  Can you please post the file /root/migrate.log?

> 
> and TL WDR 3600 v1 with firmware:
> freifunk-berlin-gonzo-1.1.0-rc1-c8948db-tl-wdr3600-v1-sysupgrade.bin

Again, what was the version installed before?  Can you please post the file /root/migrate.log?

pmelange, 2019-06-17:
I have found to problem.

The migration script is not running.  That means that the config files are not being updated to the latest version.

If you want to do it manually, edit the file /etc/config/olsr and /etc/config/olsrd and change the library names from ```olsrd_somename.so.x.y.z``` to ```olsrd_somename``` and restart olsrd (with /etc/init.d/olsrd and /etc/init.d/olsrd6).

I am changing the title of this issue to reflect the real cause of the problem.

pmelange, 2019-06-17:
The problem is with the variable CONFIG_VERSION_NUMBER.  It needs to be changed.  It will be changed in rc2 to "1.1.0-rc2-Gonzo" as this is the only way to get the name "Gonzo" to show up anywhere in LuCI or the shell banner and still work with migration.

Eventually, it will be "1.1.0-Gonzo" when we release.

It doesn't necessarily conform to "semantic versions" but who wants to conform anyhow?

CONFIG_VERSION_NICK has been dropped upstream, which was the place where we put things like "Hedy" and "Kathleen".  https://lists.infradead.org/pipermail/openwrt-adm/2017-October/000636.html

strassburger, 2019-06-17:
Thanks for the really fast digging and analyzing.
Just to complete  the information:

Base before update:
Freifunk Berlin Hedy 1.0.3 v1.0.3 / LuCI lede-17.01 branch (git-19.055.62634-0cc62b4) 

the content of the migration.log:
Migration is running on a Freifunk Berlin system
normal migration within Release ...

Tried your hints of doing it manually without success. 
It is just lack of knowledge on my side
Sorry I am not feeling home  in openwrt...
It will be fine for me if you have a look at the problem and a first solution in Gonzo xxx rc2
Turning back to Hedy 1.0.3 now for work.
Would be lucky to help with an try on rc2

Thanks again
Rainer

SvenRoederer, 2019-06-18:
Please see https://github.com/freifunk-berlin/firmware/issues/515 regarding the naming-options. What about using

> DISTRIB_ID='Freifunk Berlin'
DISTRIB_RELEASE='Gonzo'
DISTRIB_REVISION='1.1.0-rc1' <-- filled by Makefile from Git-tag

This might require switching the migration script to DISTRIB_REVISION

pmelange, 2019-06-18:
That is exactly what I want, but how do I get those values into /etc/openwrt_release?  The options I have are
```
CONFIG_VERSION_DIST="Freifunk Berlin"
CONFIG_VERSION_HWREV=""
CONFIG_VERSION_MANUFACTURER=""
CONFIG_VERSION_NUMBER="1.1.0-rc2-Gonzo"
CONFIG_VERSION_CODE=""
CONFIG_VERSION_PRODUCT=""
CONFIG_VERSION_REPO="http://buildbot.berlin.freifunk.net/buildbot/stable/%V/%T/packages"
```

These values feed into /etc/openwrt_release.  And since NICK was dropped upstream, we don't have it either.

https://lists.infradead.org/pipermail/openwrt-adm/2017-October/000636.html

SvenRoederer, 2019-06-19:
> DISTRIB_REVISION='1.1.0-rc1' <-- filled by Makefile from Git-tag

and the migration-script need to be changed to check the value of DISTRIB_REVISION in place of DISTRIB_RELEASE (https://github.com/freifunk-berlin/firmware-packages/blob/151711fd5ab571d85fcad8e44d6e291b386475c2/utils/freifunk-berlin-migration/uci-defaults/freifunk-berlin-01-migration.sh#L15)

strassburger, 2019-06-26:
new test with gonzo-1.1.0-rc2
Updates -migration scripts- runs fine.

Base before update:
Freifunk Berlin Hedy 1.0.3 v1.0.3 / LuCI lede-17.01 branch (git-19.055.62634-0cc62b4)

on GI-AR150 with firmware:
freifunk-berlin-1.1.0-rc2-gonzo-c544521-gl-ar150-sysupgrade.bin

TL WDR 3600 v1 with firmware:
freifunk-berlin-1.1.0-rc2-gonzo-c544521-tl-wdr3600-v1-sysupgrade.bin

Think we can close this Issue

closed 2019-06-26 by pmelange