freifunk-berlin/firmware #712: add support for Fritzbox 7320 / 7330SL (also marketed as 1&1 HomeServer)

matthias-klose, 2019-06-19:
A berlin-based freifunker was so kind as to compile an image for [avm fritzbox 7320](https://openwrt.org/toh/avm/avm_fritz_box_7320) which boots and does mesh. 

There are plenty of these boxes sold for 1-3 EUR at auctioning sites since some time ago the market was flooded with them as freebie routers by german DSL providers. 

A standard sysupgrade-image can be flashed using the standard ftp-method described on the OpenWrt-Wiki. These routers are limited in specs (400 mhz / 16 mb flash / 64 mb ram / 2.4 ghz) but delivered for a total price that is lower than 10 EUR could provide an affordable platform for some Berlin Freifunk spots.

There are some bugs to fix, as no LEDs are lit and probably one of the LAN ports will never work, but still it would be great to have the device in the alpha/beta-section or our firmware image table and maybe laying the foundation to support some of the popular AVM Fritzboxes.

SvenRoederer, 2019-06-19:
Probably @RobinBrandt87 can provide some commits, as he seems to have something ready (https://lists.berlin.freifunk.net/pipermail/berlin/2019-June/039463.html).

Not sure if these boards will be build on the buildbots too or just become a "quiet" target as brcm27xx (RPi)

everloop2, 2019-06-20:
Appreciate it. Have a 7330SL for testing.

SMP should be possible
-> https://forum.openwrt.org/t/how-can-we-make-the-lantiq-xrx200-devices-faster/9724
got SMP working for Fritz7312 -> 2x393Mhz
-> https://openwrt.org/toh/avm/fritz.box.wlan.7312
SMP and VoIP (VMMC) enabled at same time not possible
-> https://forum.archive.openwrt.org/viewtopic.php?id=62696&p=4

2nd Lan (Gigabit port) was not able to get working (1st port is ar8030 [whoose quirks has been fixed](https://openwrt.org/toh/avm/fritz.box.wlan.7312#openwrt_fixes), 2nd port ar8035)

maybe this helps:
-> https://openwrt-devel.openwrt.narkive.com/l08xHWo0/patch-atheros-ar8035-a-support

get LEDs working should be quite easy
-> /firmware/openwrt/target/linux/lantiq/files-4.14/arch/mips/boot/dts/FRITZ7320.dts

	aliases {
		led-boot = &power;
		led-failsafe = &power;
		led-running = &power;
		led-upgrade = &power;

		led-internet = &info_green;
		led-dsl = &power;
		led-wifi = &wlan;
	};

everloop2, 2019-06-29:
tunk test build - notunnel, SMP enabled

https://wiki.freifunk-cottbus.de/firmware:downloads -> [FritzBox7320/7330 SMP minimal Freifunk Berlin (dev-sam0815-1f70158-trunk) sysupgrade, 29.06.2019](https://wiki.freifunk-cottbus.de/_media/firmware:fritzbox7330:2019.06.29_minimal_smp_freifunk-berlin-dev-sam0815-1f70158-avm_fritz7320-sysupgrade.tar.gz)

feeds.conf:
src-git packages https://git.openwrt.org/feed/packages.git
src-git luci https://git.openwrt.org/project/luci.git
src-git routing https://git.openwrt.org/feed/routing.git
src-git freifunk https://github.com/SvenRoederer/freifunk_openwrt-packages.git
#src-git packages_berlin https://github.com/freifunk-berlin/firmware-packages.git
src-git packages_berlin https://github.com/freifunk-berlin/firmware-packages.git;SAm0815_experimental
src-git gluon https://github.com/freifunk-gluon/packages.git

config.mk
OPENWRT_SRC=https://git.openwrt.org/openwrt/openwrt.git

[2019.06.29_dmesg_minimal_smp.txt](https://github.com/freifunk-berlin/firmware/files/3341906/2019.06.29_dmesg_minimal_smp.txt)

LEDs work if set e.g. in LuCI -> System -> LED:

	-> set wifi to "default state" + trigger: "Wi-Fi on (phy0radio)" -> if wifi off -> WLAN-LED -> off

	-> add power + set to "default state"

matthias-klose, 2019-06-29:
> tunk test build - notunnel, SMP enabled

much appreciated! flashing right away.

matthias-klose, 2019-07-01:
compiled a berlin master tunneldigger myself with smp dual-core activated and it runs stable. thank you. trying your LED fixes now.

everloop2, 2019-07-04:
https://forum.openwrt.org/t/2nd-lan-port-lan2-on-avm-7330-7320-does-not-work/28895

SvenRoederer, 2019-12-09:
@kls0e @everloop2 can you give an update on the status of this target?

SvenRoederer, 2019-12-09:
There is also a similar request for Gluon: https://github.com/freifunk-gluon/gluon/pull/1805

kls0e, 2019-12-10:
status quo: 

> compiled a berlin master tunneldigger myself with smp dual-core activated and it runs stable. thank you. trying your LED fixes now. EDIT: LEDs working when configured accordingly via LuCI.
> 
> Ad-Hoc mesh working. AP working. LAN1 Gbit working. Dual-Core working. Tunneldigger working. ADSL drivers not tested. TAE phone ports not working / LAN2 not working, probably never will?

SvenRoederer, 2019-12-14:
Another Gluon-PR https://github.com/freifunk-gluon/gluon/pull/1890

From the test-report you can see, that 802.11s seems also to work.
If anyone will make a PR, I'm fine with merging. Probably not a common Target, but there are active users.
Not sure, if we will get buildbot-support for this ...

SvenRoederer, 2020-04-20:
Gluon reports some issues with the LAN-port of the 7330 (https://github.com/freifunk-gluon/gluon/issues/1943). Does it also apply to the 7330SL?

everloop2, 2020-05-08:
FritzBox7320 and FritzBox7330 have different NICs.

https://lists.berlin.freifunk.net/pipermail/berlin/2020-May/040483.html

* FritzBox7320
https://openwrt.org/toh/avm/avm_fritz_box_7320

CPU:		Lantiq Xway ARX188 PSB 50812 EL (MIPS 34Kc) 2x393 MHz
Flash:		16MB
RAM:		64MB
Wireless:	2,4GHz Atheros AR9227 b/g/n
Ethernet:	1x1000Mbit Atheros AR8035, 1x100MBit Atheros AR8030-A
xDSL:		ADSL2+
USB:		2x
Power:		12V 1.0A

- LAN 1	(Lantiq PEF7071V V1.3) > bugged (only works if connected at boot)
- LAN 2	(Lantiq PEF7071V V1.3) > not working

https://github.com/freifunk-gluon/gluon/issues/1943

https://forum.openwrt.org/t/2nd-lan-port-lan2-on-avm-7330-7320-does-not-work/28895/14

https://github.com/freifunk-berlin/firmware/issues/712

* FritzBox7330SL
https://openwrt.org/toh/avm/fritz.box.wlan.7330

CPU:		Lantiq Xway ARX188 PSB 50812 EL (MIPS 34Kc) 2x393 MHz
Flash:		16MB
RAM:		64MB
Wireless:	2,4GHz Atheros AR9227 b/g/n
Ethernet:	1x1000Mbit Atheros AR8035, 1x100MBit Atheros AR8030-A
xDSL:		ADSL2+
USB:		2x
Power:		12V 1.0A

+ LAN1 (Atheros AR8035 1Gbit) > working
- LAN2 (Atheros AR8030-A, SK4P7.4D 1242 CN, 10/100 PHY) > not working

SvenRoederer, 2020-05-08:
@everloop2  Thanks for looking this up.

So we very likely will not add 7320, but the 7330SL seems good.

everloop2, 2020-07-27:
> @everloop2 Thanks for looking this up.
> 
> So we very likely will not add 7320, but the 7330SL seems good.

Yes, still no fix for 7320 LAN connect bug - Gluon did not add, if we add we have to mention LAN1 quirks.

7330SL seems good except dead 2nd LAN.

SvenRoederer, 2020-07-28:
Somebody is willing to create a PR for the 7330SL?

kls0e, 2020-07-28:
I don‘t know, people don‘t want it because LAN2 is defunct and LAN1 only works if the port is connected at boot time so maybe we will not support it officially.

SvenRoederer, 2020-07-28:
>  LAN1 only works if the port is connected at boot time

isn't this only valid for 7320, as mentioned by @everloop2 ?

having the LAN2 defunct at all on the 7330SL seems acceptable, as we have non-functioning SFP-ports on most boards

SvenRoederer, 2020-07-31:
but as you opened the issue, you are free to close it, as it turns out the hardware-support is not in an acceptable state.

kls0e, 2020-08-01:
@SvenRoederer okay, i will kls0e the issue for now. thanks for caring tho :)

closed 2020-08-01 by kls0e

everloop2, 2020-08-09:
> > LAN1 only works if the port is connected at boot time
> 
> isn't this only valid for 7320, as mentioned by @everloop2 ?
> 
> having the LAN2 defunct at all on the 7330SL seems acceptable, as we have non-functioning SFP-ports on most boards

gluon did not add **FritzBox7320** cause of LAN1 bugged (has to be connected at boot time) and LAN2 is dead.

https://github.com/freifunk-gluon/gluon/issues/1943#issuecomment-610943676

https://bugs.openwrt.org/index.php?do=details&task_id=2240&string=fritz&type%5B0%5D=&sev%5B0%5D=&pri%5B0%5D=&due%5B0%5D=&reported%5B0%5D=&cat%5B0%5D=&status%5B0%5D=open&percent%5B0%5D=&opened=&dev=&closed=&duedatefrom=&duedateto=&changedfrom=&changedto=&openedfrom=&openedto=&closedfrom=&closedto=

**FritzBox7330SL** LAN1 (Atheros 8035 1Gbit) is working, LAN2 (Atheros 8030 100MBit) is "dead"

so FritzBox7330SL is in a working state as an AP

SvenRoederer, 2020-08-18:
@everloop2 that for this closing review, but @kls0e seems not interested in these boards anymore.