freifunk-berlin/firmware #715: fix "bashish

SvenRoederer, 2019-06-30:


SvenRoederer, 2019-12-28:
The syntax "[[" is also a bash-extension (https://stackoverflow.com/questions/669452/is-double-square-brackets-preferable-over-single-square-brackets-in-ba) not supported in ash. So scripts using this syntax will not work as expected.

works in BASH:
> bash$ b="test 1234"
> bash$ [[ "test 1234" = $b ]] && echo strings are equal
> strings are equal

fails in Busybox/ASH:
> ash$ b="test 1234"
> ash$ [[ "test 1234" = $b ]] && echo strings are equal
> -ash: 1234: unknown operand

SvenRoederer, 2019-12-29:
PR https://github.com/openwrt/openwrt/pull/2647 tries to remove some bashism in OpenWrt

Akira25, 2020-04-20:
The reason for this issue remains open is the unmerged upstream commit to OpenWRT, right?

SvenRoederer, 2020-04-20:
Most of the referenced upstream commits have been merged already on 2020-01-26. The last one I just checked and resubmitted again. I assume it will be merged soon.
@Akira25 thanks for the ping.

But I think our repo still contains the bashism ...

Akira25, 2020-04-20:
I did some grep-stuff on recent checkouts of the firmware and the firmware-packages-repo. There where matches in the packages repo only:

`grep -rl "\[\[" . | grep .sh`
```
./utils/freifunk-berlin-migration/uci-defaults/freifunk-berlin-01-migration.sh
./utils/freifunk-berlin-migration/files/lib/functions/semver.sh
```

`grep -rl "==" . | grep .sh`
```
./utils/freifunk-berlin-wizard-backend/files/usr/lib/ffwizard.d/30-batman-adv.sh
./utils/freifunk-berlin-wizard-backend/files/usr/lib/ffwizard.d/20-network.sh
./utils/freifunk-berlin-wizard-backend/files/usr/lib/ffwizard.d/30-olsrd.sh   
./utils/freifunk-berlin-wizard-backend/files/usr/lib/ffwizard.d/15-wireless.sh
./utils/freifunk-berlin-wizard-backend/files/usr/lib/ffwizard.d/60-dhcp.sh   
./utils/freifunk-berlin-wizard-backend/files/usr/lib/ffwizard.d/25-tunnels.sh
./utils/freifunk-berlin-wizard-backend/files/usr/lib/ffwizard.d/30-olsrd2.sh  
./utils/freifunk-berlin-migration/uci-defaults/freifunk-berlin-01-migration.sh
./utils/freifunk-berlin-migration/files/lib/functions/semver.sh
./utils/luci-app-ffwizard-berlin/luasrc/model/cbi/freifunk/assistent/shareInternet.lua
./addons/freifunk-berlin-lib-guard/files/lib/functions/guard.sh
./addons/freifunk-berlin-configtools/root/usr/share/freifunk-berlin/bandwidth-change.sh
./addons/freifunk-berlin-configtools/root/usr/share/freifunk-berlin/sharenet-switch.sh
```

In the firmware-repo matches only did happen in patch files. But i assume, that they are fine, as the build environment uses bash, right?

SvenRoederer, 2020-04-20:
Correct host-commands of the build-scripts are executed by the host-bash. But the packages-code will run on the OpenWrt-Router with ash.
Let's forget about the "freifunk-berlin-wizard" as this code is deprecated and nobody maintaining it.