freifunk-berlin/firmware #718: /daily/upstream-1907 luci did not work

RobinBrandt87, 2019-07-28:
The package "luci-lib-ipkg" was split from "luci-base" in openwrt/luci@4bbe325. 
This problem was  solved in https://github.com/freifunk-berlin/firmware-packages/commit/c76c35b2b0bb0049906ea8171783fe9ca4e25938. We can fix this by creating a new branch of the packages for the 1907 an change it. Or use the version from SAm0815, but i don't know other changes in this branch.

pmelange, 2019-07-29:
Temporary workaround on a freshly installed daily/upstream_1907 image (don't forget to connect the WAN port):

```
echo "nameserver 8.8.8.8" > /etc/resolv.conf
cd /tmp
wget https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/luci/luci-lib-ipkg_git-19.204.23774-bd44189-1_all.ipk
opkg install /tmp/luci-lib-ipkg_git-19.204.23774-bd44189-1_all.ipk
```

Yes, I agree the luci-lib-ipkg package should be included in these builds.  I have privately written @SvenRoederer asking him to do this.  For a reason I didn't quite understand, he didn't want to.

SvenRoederer, 2019-08-03:
See also https://github.com/freifunk-berlin/firmware/issues/668

As mentioned in this private conversation, the daily/upstream-1907 branch of firmware is using the regular master-branch of our packages-feed. This master-branch is not having a dependency of luci-lib-ipkg. So the solution seems to make a new 1907-branch of or packages-feed and include the missing dependency. 
But I don't have the resources to maintain another branch and keep in sync with master. One reason are the constant rebase-conflicts from package-version-number (see PR https://github.com/freifunk-berlin/firmware-packages/pull/188 for my idea to work around this).
I sugget:
* getting rid of using luci-lib-ipkg
* create a branch for 1907 any maintain it

SvenRoederer, 2019-08-03:
> echo "nameserver 8.8.8.8" > /etc/resolv.conf
> cd /tmp
> wget https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/luci/luci-lib-ipkg_git-19.204.23774-bd44189-1_all.ipk
> opkg install /tmp/luci-lib-ipkg_git-19.204.23774-bd44189-1_all.ipk

You can also do:

> echo "nameserver 8.8.8.8" > /etc/resolv.conf
> opkg update
> opkg install luci-lib-ipkg

SvenRoederer, 2019-08-03:
@RobinBrandt87 wrote:

> Or use the version from SAm0815, but i don't know other changes in this branch.

I strongly do not recommend using SAm0815_experimental branch as base for other branches.

SvenRoederer, 2019-08-09:
I integrated PR #714 into the upstream-1907 branch. Based on this I put https://github.com/freifunk-berlin/firmware-packages/pull/158 via "git show" as a patch into the build. 
Hope this works with less maintenance.

SvenRoederer, 2019-12-16:
After switching the firmware-master to openwrt-1907 we officially updated the dependencies to include "luci-lib-ipkg" (https://github.com/freifunk-berlin/firmware-packages/commit/582fec0255e01927a3b4f008b0030f98f7fafa82).
So this issue is fixed.

closed 2019-12-16 by SvenRoederer