freifunk-berlin/firmware #719: /daily/upstream-1907 olsr did not work correctly

RobinBrandt87, 2019-07-28:
since https://github.com/openwrt-routing/packages/commit/20b9ab3ceae3658046cfe6e09a699071d512dfa8#diff-85f231f1dd3b157304ac03774a9ef3fa
openwrt use the olsr version 0.9.7. by testing the new ath79 target becomes the problem that the routing to other services/user/router in the network did not work. The smartgatway works fine Internet is OK but the rest. Changing the olsr version to 0.9.6.2 solve this problem.

pmelange, 2019-07-29:
Can you please be a bit more specific with you test cases.  I would like to try to reproduce it.

The reason that I ask is that the changes from 0.9.6.2 to 0.9.7 impact the parts of the code that we use minimally.

See the diff here: https://github.com/OLSR/olsrd/compare/v0.9.6.2...v0.9.7

I have set up nodes with 0.9.7 and haven't had any routing problems.... yet.

SvenRoederer, 2019-12-29:
@RobinBrandt87 our master is now based on OpenWrt-19.07. there might have been come changes to olsrd also.
Can you retest the problem?

SvenRoederer, 2020-10-18:
Closing, as there is not feedback since >12 months.

@RobinBrandt87 feel free to reopen with additional information.

closed 2020-10-18 by SvenRoederer