freifunk-berlin/firmware #720: RFC: More Busybox Options

pmelange, 2019-07-30:
I have been looking though the options for busybox and would like to recommend turning on the following compilation flags.  (I am just listing the "CONFIG_BUSYBOX_DEFAULT_*" options, but for each one there is also a corresponding "CONFIG_BUSBOX_CONFIG_*" option).

CONFIG_BUSYBOX_DEFAULT_FEATURE_TOP_INTERACTIVE
CONFIG_BUSYBOX_DEFAULT_FEATURE_TOP_SMP_CPU
CONFIG_BUSYBOX_DEFAULT_FEATURE_TOPMEM
CONFIG_BUSYBOX_DEFAULT_FEATURE_VI_UNDO
CONFIG_BUSYBOX_DEFAULT_FEATURE_LESS_REGEXP
CONFIG_BUSYBOX_DEFAULT_FEATURE_LESS_LINENUMS
CONFIG_BUSYBOX_DEFAULT_WATCH
CONFIG_BUSYBOX_DEFAULT_DIFF

These would be nice, but IMHO not super necessary
CONFIG_BUSYBOX_DEFAULT_FEATURE_WGET_HTTPS
CONFIG_BUSYBOX_DEFAULT_FEATURE_WGET_OPENSSL

What do you think about these options?  Are there any others that you would like as well?

SvenRoederer, 2019-07-30:
> CONFIG_BUSYBOX_CONFIG_DIFF

opkg install diff-utils?

> CONFIG_BUSYBOX_CONFIG_FEATURE_WGET_HTTPS
> CONFIG_BUSYBOX_CONFIG_FEATURE_WGET_OPENSSL

As far as I know, we are using the ustream-module for https already.  So what's advantage of enabling the SSL stuff inside busybox too?

All of this options will have an effect of the total binary-size, right?

pmelange, 2019-07-30:
> > CONFIG_BUSYBOX_CONFIG_DIFF
> 
> opkg install diff-utils?

Diff is, in my opinion, a tool which should always be installed.  Sure, it could be installed after installation, but having it in the squashfs is even better.

> 
> > CONFIG_BUSYBOX_CONFIG_FEATURE_WGET_HTTPS
> > CONFIG_BUSYBOX_CONFIG_FEATURE_WGET_OPENSSL
> 
> As far as I know, we are using the ustream-module for https already. So what's advantage of enabling the SSL stuff inside busybox too?

Might be.  I haven't used https with wget in ages since I always thought it didn't work.  But now I just tried it out, and it worked find.  OK.  So those two option are not necessary.

> 
> All of this options will have an effect of the total binary-size, right?

Since we are not supporting the 4MB targets any more, then the few more bytes needed to get these options are, IMHO, irrelevant.

db4rne, 2019-07-30:
@pmelange I feel like it might be better to talk numbers here: e.g.: why would we want openssl support and how much space are we talking about? Don't we need openssl then as well? As far as I remember, openssl is rather big.

pmelange, 2019-07-30:
Please forget the whole WGET_HTTPS and WGET_OPENSSL.  I though that the current wget didn't work with https requests.  But it does.

SvenRoederer, 2019-07-30:
The mayor part of the system is ready to support SSL, also opkg can support this since long time, see #148, #492

SvenRoederer, 2019-08-06:
I'm not sure if we should break with 4MB if there is no real need. 
For less and vi(m) the full featured software from the packages feed can be used as needed. There is probably also a top-binary

SvenRoederer, 2019-11-21:
We can also disable these options on the 4MB board explicitly via their .config.
But do we want to introduce a different set of available commands on these boards? This might cause problems when making scripts, which rely on these commands.