freifunk-berlin/firmware #721: TP-Link CPE210 v2 1cfbb77_sysupgrade firmware bugs

everloop2, 2019-08-10:
2019.08.05_freifunk-berlin-development-1cfbb77-cpe210-v2-sysupgrade.bin

1. LuCI firmware upgrade: "Das hochgeladene Firmware-Image hat ein nicht unterstütztes Format. Stellen Sie sicher dass Sie das generische Format für Ihre Platform gewählt haben."
Did a firmware update via tftp -> working

2. LuCI bug:
```
/usr/lib/lua/luci/dispatcher.lua:498: bad argument #1 to 'pairs' (table expected, got nil)
stack traceback:
[C]: in function 'pairs'
/usr/lib/lua/luci/dispatcher.lua:498: in function 'createtree'
/usr/lib/lua/luci/dispatcher.lua:207: in function 'dispatch'
/usr/lib/lua/luci/dispatcher.lua:99: in function </usr/lib/lua/luci/dispatcher.lua:98>
```
3. 2019.08.05_freifunk-berlin-development-1cfbb77-cpe210-v2-factory.bin -> master branch
LuCI working, wizard + reboot working, wifi working

SvenRoederer, 2019-08-10:
problem 1& 3 might be related to each other, but problem 2 seems unrelated. 

Can you open a separate issue for probelm 2? It's also not clear to me, in what situation problem 2 occurred.

everloop2, 2019-08-20:
tried -> http://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx-generic/1541/default/freifunk-berlin-development-1cfbb77-cpe210-v2-sysupgrade.bin (master branch 2019.08.19)

-> LuCI firmware upgrade:

> Das hochgeladene Firmware-Image hat ein nicht unterstütztes Format. Stellen Sie sicher dass Sie das generische Format für Ihre Platform gewählt haben.
> "Upgrade erzwingen" auswählen um die Abbilddatei auch dann zu schreiben, wenn die Formatüberprüfung fehlschlägt. Diese Option nur benutzen wenn
> das Abbild korrekt und für dieses Gerät bestimmt ist!

-> workaround: Force upgrade -> freifunk wizard + reboot & wifi -> working :D

-> 2. LuCI bug seems to be gone -> this why could not test: wizard + reboot & wifi

-> miracles in future -> wifi will be broken :[
-> /issues/722

SvenRoederer, 2019-08-29:
> tried -> http://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx-generic/1541/default/freifunk-berlin-development-1cfbb77-cpe210-v2-sysupgrade.bin (master branch 2019.08.19)

this build is freifunk-berlin-master (based on OpenWrt 18.06 of 3.6.2019) and quite outdated (deprecated ar71xx-target). 
* Have you tried "upstream master" or "upstream-1907" branch?
* What's the firmware you upgraded from?

everloop2, 2019-12-05:
Some time passed - i'll close this one due don't think can repuduce it.

closed 2019-12-05 by everloop2