freifunk-berlin/firmware #722: TP-Link CPE210 Ver.3.0 not working

Mitzlaff, 2019-08-14:
### Bug report

**What is the problem?**
WLAN does not work. 

When using LAN connection I see this:

[image]

**What is the expected behaviour?**
SSID showing up when scanning for configured SSID name (freifunk-backyard)

**Firmware Version:**
freifunk-berlin-dev-daily-6431131-tplink_cpe210-v3-sysupgrade.bin

**Site Configuration:**
default, just using Luci

**Custom patches:**
none

SvenRoederer, 2019-08-14:
Does disabling the Mesh-link (intern-ch13.freifunk.net) fixes it?

everloop2, 2019-08-20:
see: https://bugs.openwrt.org/index.php?do=details&task_id=2333

https://github.com/openwrt/openwrt/pull/1951#issuecomment-512795229

2-1 week ago searched for wifi bug - other devices using similar hardware seems to affected too

psyborg55, 2019-10-02:
> -> wifi bug hint: seen somebody mentioned in CPE210 bug threads different irqs

doesn't appear to be a problem, at least according to bootlog https://github.com/openwrt/openwrt/pull/2468#issuecomment-537549406

AR9344 and AR9342 work fine with this irq setup

everloop2, 2019-11-16:
seems to be fixed:

mac80211 ath9k: force QCA953x clock to 25MHz

https://github.com/openwrt/openwrt/commit/4c6fe32468bc60cc25a8c298498c0be3c73e7378

> The QCA953x only supports 25 MHz refclk, however some OEMs set an
> invalid bootstrap value for the REF_CLK option, which would break the
> clock detection in ath9k.
> 
> Force the QCA953x refclk to 25MHz in ath9k, as this is (according to the
> datasheet) the only valid frequency.

> Signed-off-by: David Bauer <mail@david-bauer.net>

did not test it yet.

SvenRoederer, 2019-11-26:
> **Firmware Version:**
> freifunk-berlin-dev-daily-6431131-tplink_cpe210-v3-sysupgrade.bin
> 
> ****

For the records: this was the first build based on openwrt-master using the sth79-target for the CPE210-devices.

Mitzlaff, 2019-11-27:
@SvenRoederer is the openwrt fix https://github.com/openwrt/openwrt/commit/4c6fe32468bc60cc25a8c298498c0be3c73e7378 already in the freifunk firmware? Then it would be easy for me to test it.

SvenRoederer, 2019-11-27:
@Mitzlaff  the current build of branch "daily/upstream-master" contains all openwrt-commit till 23.Nov. So the mentioned one is also included.

Looking forward to your test-results.

everloop2, 2019-12-05:
on 21.11.2019 did an OpenWrt trunk test Build:

Powered by LuCI Master (git-19.324.23065-4fe16b1) / OpenWrt SNAPSHOT r11562-08d9c95417

wifi working:

clients can connect to TP-Link CPE210 Ver: 2.0 and

CPE210 is connected via wifi to a Freifunk Box -> internet working

SvenRoederer, 2019-12-05:
@everloop2  do you know the exact filename of the image you flashed? the filename also contains the githash, to back-reference to exact code used for this build.

But thanks for testing

everloop2, 2019-12-06:
unfortuneatly -> openwrt-ath79-generic-tplink_cpe210-v2-squashfs-sysupgrade.bin -> o.O

hope/think its right:

https://github.com/openwrt/openwrt/commits/317e98a5a2 -> Commits on Nov 21, 2019

Kernel Log
[    0.000000] Linux version 4.19.84 (human@rechenknecht37) (gcc version 8.3.0 (OpenWrt GCC 8.3.0 r11561-**317e98a5a2**)) #0 Thu Nov 21 11:55:50 2019
```
[    0.000000] Linux version 4.19.84 (human@rechenknecht37) (gcc version 8.3.0 (OpenWrt GCC 8.3.0 r11561-317e98a5a2)) #0 Thu Nov 21 11:55:50 2019
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019374 (MIPS 24Kc)
[    0.000000] MIPS: machine is TP-Link CPE210 v2
[    0.000000] SoC: Qualcomm Atheros QCA9533 ver 2 rev 0
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 04000000 @ 00000000 (usable)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] On node 0 totalpages: 16384
[    0.000000]   Normal zone: 144 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 16384 pages, LIFO batch:3
[    0.000000] random: get_random_bytes called from start_kernel+0x98/0x4a8 with crng_init=0
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 16240
[    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] Memory: 57580K/65536K available (4273K kernel code, 175K rwdata, 1012K rodata, 1188K init, 206K bss, 7956K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 51
[    0.000000] CPU clock: 650.000 MHz
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 5880801374 ns
[    0.000010] sched_clock: 32 bits at 325MHz, resolution 3ns, wraps every 6607641598ns
[    0.008278] Calibrating delay loop... 432.53 BogoMIPS (lpj=2162688)
[    0.074889] pid_max: default: 32768 minimum: 301
[    0.080036] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.087012] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.100222] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.110680] futex hash table entries: 256 (order: -1, 3072 bytes)
[    0.117304] pinctrl core: initialized pinctrl subsystem
[    0.124132] NET: Registered protocol family 16
[    0.168056] clocksource: Switched to clocksource MIPS
[    0.174965] NET: Registered protocol family 2
[    0.180637] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 4096 bytes)
[    0.188887] TCP established hash table entries: 1024 (order: 0, 4096 bytes)
[    0.196249] TCP bind hash table entries: 1024 (order: 0, 4096 bytes)
[    0.203004] TCP: Hash tables configured (established 1024 bind 1024)
[    0.209911] UDP hash table entries: 256 (order: 0, 4096 bytes)
[    0.216098] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes)
[    0.223299] NET: Registered protocol family 1
[    0.227939] PCI: CLS 0 bytes, default 32
[    0.232204] Crashlog allocated RAM at address 0x3f00000
[    0.239886] workingset: timestamp_bits=14 max_order=14 bucket_order=0
[    0.254776] squashfs: version 4.0 (2009/01/31) Phillip Lougher
[    0.261008] jffs2: version 2.2 (NAND) (SUMMARY) (LZMA) (RTIME) (CMODE_PRIORITY) (c) 2001-2006 Red Hat, Inc.
[    0.289410] io scheduler noop registered
[    0.293550] io scheduler deadline registered (default)
[    0.301570] pinctrl-single 1804002c.pinmux: 576 pins, size 72
[    0.308922] Serial: 8250/16550 driver, 1 ports, IRQ sharing disabled
[... 91 lines cut ...]
[   30.495521] IPv6: ADDRCONF(NETDEV_UP): wlan0-1: link is not ready
[   30.561984] br-lan: port 2(wlan0-1) entered blocking state
[   30.567812] br-lan: port 2(wlan0-1) entered disabled state
[   30.574108] device wlan0-1 entered promiscuous mode
[   31.738748] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   33.188526] wlan0: authenticate with 7c:4f:b5:31:67:f7
[   33.207298] wlan0: send auth to 7c:4f:b5:31:67:f7 (try 1/3)
[   33.218827] wlan0: authenticated
[   33.288195] wlan0: associate with 7c:4f:b5:31:67:f7 (try 1/3)
[   33.301571] wlan0: RX AssocResp from 7c:4f:b5:31:67:f7 (capab=0x421 status=0 aid=1)
[   33.309891] wlan0: associated
[   33.326788] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   33.361095] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0-1: link becomes ready
[   33.368386] br-lan: port 2(wlan0-1) entered blocking state
[   33.374101] br-lan: port 2(wlan0-1) entered forwarding state
[  942.422119] eth0: link up (10Mbps/Half duplex)
[  963.222450] eth0: link up (100Mbps/Full duplex)
[ 1049.567857] eth0: link up (10Mbps/Half duplex)
[ 1063.106729] eth0: link up (100Mbps/Full duplex)
[ 1086.038190] eth0: link down
[ 1086.042765] br-lan: port 1(eth0) entered disabled state
[ 1087.081039] eth0: link up (100Mbps/Full duplex)
[ 1087.085856] br-lan: port 1(eth0) entered blocking state
[ 1087.091352] br-lan: port 1(eth0) entered forwarding state
```

SvenRoederer, 2019-12-07:
Ahh, you used a vanilla OpenWrt for your testing. which will match with our code d98f5205b2ff30ec19ee3d0d9e28c96c9a9e59a5

everloop2, 2019-12-09:
today tried to build: daily/upstream-master

```
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-mod-failsafe:
 *      luci-base
 * opkg_install_cmd: Cannot install package luci-mod-failsafe.
```
think patch/commits https://github.com/freifunk-berlin/firmware/commit/3237490c302bb6ac9d75793f16fd5db02b76074b are missing

at master: /patches/packages/luci/0202-split-luci-base_PR2817.patch
at daily/upstream-master:  /patches/packages/luci/0001-luci-PR2817_split_luci-base.patch

SvenRoederer, 2019-12-09:
> today tried to build: daily/upstream-master

let's agree on commit 39e7543 of branch daily/upstream-master

> ```
> Collected errors:
>  * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-mod-failsafe:
>  *      luci-base
>  * opkg_install_cmd: Cannot install package luci-mod-failsafe.
> ```
> 
> think patch/commits [3237490](https://github.com/freifunk-berlin/firmware/commit/3237490c302bb6ac9d75793f16fd5db02b76074b) are missing

As you can see, the PR2817 patch splits the current luci-base package into luci-lib-base and luci-base, which depends on luci-lib-base. Without the patch the luci-base package is there, as it's in the original luci-feed. With the patch both packages are there and fulfill all dependencies.
There must be another cause of the problem. 
Have you additional logs?
can you publish the code you are using?

everloop2, 2019-12-10:
trying again fresh build (10.12.2019 14:50):

```
umask 0022
git clone https://github.com/freifunk-berlin/firmware.git
cd firmware
git checkout daily/upstream-master
/firmware/profiles/ath79-generic.profiles -> removed everything except: tplink_cpe210-v2
make -j2 V=s TARGET=ath79-generic PACKAGES_LIST_DEFAULT=notunnel
```

everloop2, 2019-12-10:
```
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-mod-failsafe:
 *      luci-base
 * opkg_install_cmd: Cannot install package luci-mod-failsafe.

make[3]: *** [package/Makefile:68: package/install] Error 255

make[2]: *** [package/Makefile:108: firmware/openwrt/staging_dir/target-mips_24kc_musl/stamp/.package_install] Error 2

/firmware/openwrt/include/toplevel.mk:227: world] Fehler 2
make[1]: Verzeichnis „/firmware/openwrt“ wird verlassen
make: *** [Makefile:189: .stamp-compiled] Fehler 2

```

SvenRoederer, 2019-12-10:
> trying again fresh build (10.12.2019 14:50):
> git checkout daily/upstream-master

--> https://github.com/freifunk-berlin/firmware/commit/39e75437746c259164972aa9525b0e562fb070ba (to browser the directory-tree)

you can verify with `git status`

---

@everloop2 I assume you are affected by the problem mentioned in https://github.com/freifunk-berlin/firmware/pull/714#issuecomment-546705095 . My approach to split the patches-directory (PR #714), was sometimes causing problems on the feeds. I assume you will find in  the logs (using V=s make option) a message indicating that the luci-base Makefile will be found in `openwrt/feeds/.pc/...` . This path is of course invalid and will not give a valid package.
I did not find the real cause or a nice solution, so I switched over to use the gluon-patch-system (which is merged to master now). 
I need to switch the daily/master-upstream branch to this patch-framework to, when I've time. In the meantime you can just disable the PR2817 patch, which should make the build not fails at all.

To check for the fault you can run:
`make TARGET=ath79-generic V=s prepare 2>&1 | less`

everloop2, 2019-12-12:
`make -j2 V=s TARGET=ath79-generic PACKAGES_LIST_DEFAULT=notunnel`

shown straight at compile start:

```
rm -f .stamp-firmwares
rm -rf /firmware/openwrt/bin
rm -rf /firmware/openwrt/build_dir/target-*/*-{imagebuilder,sdk}-*
cat /firmware/configs/common.config /firmware/configs/ath79-generic.config > /firmware/openwrt/.config
# always replace CONFIG_VERSION_CODE by FW_REVISION
sed -i "/^CONFIG_VERSION_CODE=/c\CONFIG_VERSION_CODE=\"39e7543\"" /home/human/Freifunk/firmware_daily_up_master/firmware/openwrt/.config
umask 022; \
  make -C /home/human/Freifunk/firmware_daily_up_master/firmware/openwrt defconfig
make[1]: Verzeichnis „/home/human/Freifunk/firmware_daily_up_master/firmware/openwrt“ wird betreten
make[2]: Entering directory '/home/human/Freifunk/firmware_daily_up_master/firmware/openwrt'
make[2]: Leaving directory '/home/human/Freifunk/firmware_daily_up_master/firmware/openwrt'
WARNING: Makefile 'package/feeds/packages/acme/Makefile' has a dependency on 'luci-base', which does not exist
WARNING: Makefile 'package/feeds/routing/bird1-ipv4-openwrt/Makefile' has a dependency on 'luci-base', which does not exist
WARNING: Makefile 'package/feeds/routing/bird1-ipv6-openwrt/Makefile' has a dependency on 'luci-base', which does not exist
WARNING: Makefile 'package/feeds/luci/luci/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-adblock/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-advanced-reboot/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-ahcp/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-aria2/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-attendedsysupgrade/Makefile' has a dependency on 'luci-base', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-attendedsysupgrade/Makefile' has a build dependency on 'luci-base/host', which does not exist                                                                                                                                          
WARNING: Makefile 'package/feeds/luci/luci-app-banip/Makefile' has a build dependency on 'luci-base/host', which does not exist            
WARNING: Makefile 'package/feeds/luci/luci-app-bcp38/Makefile' has a build dependency on 'luci-base/host', which does not exist            
WARNING: Makefile 'package/feeds/routing/luci-app-bmx6/Makefile' has a dependency on 'luci-base', which does not exist                     
WARNING: Makefile 'package/feeds/routing/luci-app-bmx6/Makefile' has a dependency on 'luci-lib-httpclient', which does not exist           
WARNING: Makefile 'package/feeds/luci/luci-app-bmx7/Makefile' has a build dependency on 'luci-base/host', which does not exist             
WARNING: Makefile 'package/feeds/luci/luci-app-cifsd/Makefile' has a build dependency on 'luci-base/host', which does not exist            
WARNING: Makefile 'package/feeds/routing/luci-app-cjdns/Makefile' has a dependency on 'luci-base', which does not exist                    
WARNING: Makefile 'package/feeds/luci/luci-app-clamav/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-commands/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-coovachilli/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-cshark/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-dcwapd/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-ddns/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-diag-core/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-dnscrypt-proxy/Makefile' has a dependency on 'luci-lib-httpprotoutils', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-dnscrypt-proxy/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-dump1090/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-dynapoint/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-firewall/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/feeds/freifunk/luci-app-freifunk-diagnostics/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/feeds/freifunk/luci-app-freifunk-policyrouting/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/feeds/freifunk/luci-app-freifunk-widgets/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-fwknopd/Makefile' has a build dependency on 'luci-base/host', which does not exist
[... cut ...]
WARNING: Makefile 'package/feeds/luci/luci-theme-material/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-theme-openwrt/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-theme-rosy/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/feeds/packages/sqm-scripts/Makefile' has a dependency on 'luci-base', which does not exist
touch .config
[...]
```

everloop2, 2019-12-12:
> I need to switch the daily/master-upstream branch to this patch-framework to, when I've time. In the meantime you can just disable the PR2817 patch, which should make the build not fails at all.

disabled PR2817: `/firmware/patches/packages/luci/series -> #0001-luci-PR2817_split_luci-base.patch`

-> same build error

No stress - only wanted to test daily/upstream-master branch

Will try test build on master, 25Mhz patch should be included:
https://git.openwrt.org/?p=openwrt/openwrt.git;a=shortlog;h=refs/heads/openwrt-19.07;pg=1
2019-11-06 | David Bauer | mac80211 ath9k: force QCA953x clock to 25MHz

https://github.com/freifunk-berlin/firmware/commit/81aad75392ddcaf8d1e6974c816357aa06add8df
Update to HEAD of openwrt-19.07 branch (9. Nov 2019)

everloop2, 2019-12-15:
master branch working :)

`[    0.000000] Linux version 4.14.151 (human@rechenknecht37) (gcc version 7.4.0 (OpenWrt GCC 7.4.0 r10666+10-d3e11e8ad8)) #0 Fri Sep 6 08:29:50 2019
`

tested as mesh node: wizard + mesh working -> only "set-password-again-bug" (no major)

[kernelprotokoll.txt](https://github.com/freifunk-berlin/firmware/files/3964895/kernelprotokoll.txt)
[systemprotokoll.txt](https://github.com/freifunk-berlin/firmware/files/3964896/systemprotokoll.txt)

SvenRoederer, 2019-12-19:
so master-branch (openWrt-19.07 based) , right?

thanks for testing, keep us informed

everloop2, 2019-12-22:
Yep, plain master: https://github.com/freifunk-berlin/firmware/tree/master

Deactivated IPv6 everywhere: Olsrd, NetworkDevices, SystemStartup -> does not help.

-> lantiq boxes: in past (more half year ago) stumbled over a high CPU load bug -> deactivate IPv6 did help.

Vanilla OpenWrt does not seem to have high CPU load and self-reboots:

https://downloads.openwrt.org/snapshots/targets/ath79/generic/tplink_cpe210-v2-squashfs-factory.bin

SvenRoederer, 2020-06-28:
just made PR #827. Is the node  "cottbus-lausi36-2" still a CPE-v3 on ath79?

everloop2, 2020-07-21:
> just made PR #827. Is the node "cottbus-lausi36-2" still a CPE-v3 on ath79?

Was a tplink_cpe210-v2 (1xLAN).

Trying build branch cpe210 fails due gcc10 support check fails (arch linux): https://patchwork.ozlabs.org/project/openwrt/patch/20200318183943.529630-1-robimarko@gmail.com/

SvenRoederer, 2020-07-21:
Yeah, I see your build issues are only with our outdated master-branch.

The "mksquashfs"-bug mentioned is already backported to 19.07 (https://github.com/openwrt/openwrt/commit/059db419d7dfbf8e27c49b1f6b88e8170d659088). - 

Probably you want to give a build of https://github.com/SvenRoederer/freifunk-berlin-firmware/commits/nightly/upstream-1907 a chance. I can also share you the prebuild imagebuilder, but confirming build on arch linux would be also great.

everloop2, 2020-07-22:
> Yeah, I see your build issues are only with our outdated master-branch.
> 
> The "mksquashfs"-bug mentioned is already backported to 19.07 ([openwrt/openwrt@059db41](https://github.com/openwrt/openwrt/commit/059db419d7dfbf8e27c49b1f6b88e8170d659088)). -
> 
> Probably you want to give a build of https://github.com/SvenRoederer/freifunk-berlin-firmware/commits/nightly/upstream-1907 a chance. I can also share you the prebuild imagebuilder, but confirming build on arch linux would be also great.

canceled building https://github.com/freifunk-berlin/firmware/commits/openwrt-19.07.3

did a build for: https://github.com/SvenRoederer/freifunk-berlin-firmware/commits/nightly/upstream-1907 on Linux 5.7.9-1-MANJARO (x86_64)

openwrt gcc 10: internal compiler error: Segmentation fault > 1times in early build process > seems doesnt like multithreaded building (make -j4)

make -j1 > build continues > afterwards multithreaded again (make -j4) > build finished :)
(maybe `make download` or something smt related)

build: Powered by LuCI branch (git-19.082.34170-bf97eae) / Freifunk Berlin SAm0815-master 0017a2fb

- flashing & wizard working

bugs:

- wizard finished > reboot > LuCI shows no menu or login buttons
- ssh working
- mesh working > speedtest > almost max. connetion (10mbit here)
- cpu idle: ~70% > rssi-led bug > option dev 'wlan0'
- cpu idle: ~90% > option dev 'wlan0-mesh-2'

SvenRoederer, 2020-07-22:
> did a build for: https://github.com/SvenRoederer/freifunk-berlin-firmware/commits/nightly/upstream-1907 on Linux 5.7.9-1-MANJARO (x86_64)
> 
> ...
> 
> build: Powered by LuCI branch (git-19.082.34170-bf97eae) / Freifunk Berlin SAm0815-master 0017a2f
> 
Based on the version-string of your firmware you built my repos master branch (OpenWrt-19.07 of 21st June) and not on "nightly/upstream-1907" branch (recent OpenWrt-19.07). I expect this to be the cause of the build-failure you have seen.

>     * flashing & wizard working
> 

That's great to hear, so its proofed that CPE210v2 is working without wifi-issues. As per https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=141698ce8fbc52d253c9207ef46b3630a0870567 the v2 and v3 are identical I assume v3 is also working. 
So I'm looking forward to get #827 merged.

> 
> bugs:
> 
>     * wizard finished > reboot > LuCI shows no menu or login buttons
--> see issue #831
>     * ssh working
>     * mesh working > speedtest > almost max. connetion (10mbit here)
--> are these bugs? :-)
>     * cpu idle: ~70% > rssi-led bug > option dev 'wlan0'
>     * cpu idle: ~90% > option dev 'wlan0-mesh-2'
--> this references to issue #769?

SvenRoederer, 2020-07-22:
@Mitzlaff @everloop2 shall we close this?

everloop2, 2020-07-23:
> @Mitzlaff @everloop2 shall we close this?

Close, wifi has been fixed, other bugs > other issues.

yep no bugs ;)
* ssh working
* mesh working > speedtest > almost max. connetion (10mbit here)

closed 2020-07-25 by SvenRoederer