freifunk-berlin/firmware #723: Request Firmware support for Netgear EX6150v2 and R6120

captainkingcandy, 2019-08-19:
Hello,

i would like to request firmware support for two netgear devices, which are already openwrt compatible:

[https://openwrt.org/toh/hwdata/netgear/netgear_ex6150_v2](url)
[https://openwrt.org/toh/hwdata/netgear/netgear_r6120](url)

im not sure, how to add the right profile and configs. I already wrote an email to @SvenRoederer. Would be great if someone could help me out :-) 

Lukas

SvenRoederer, 2019-08-19:
This device was brought into attention by @RobinBrandt87 via https://lists.berlin.freifunk.net/pipermail/berlin/2019-August/039619.html

captainkingcandy, 2019-08-20:
yesterday i got a few hours time and tried to compile a firmware for the netgear ex6510v2.
you can find it here:

[http://buildbot.berlin.freifunk.net/buildbot/unstable/ipq40xx-generic/158/tunnel-berlin-tunneldigger/freifunk-berlin-dev-daily-1907-b49cb91-ipq40xx-netgear_ex6150v2-squashfs-factory.img](url)

maybe i will be able to test it today or tomorrow, so i can give some feedback.

SvenRoederer, 2019-08-20:
This was an easy one. Looking forward to the feedback of your test.

also adding them to the Routerlist on the firmware-page would be nice (https://wiki.freifunk.net/Berlin:Firmware#WLAN-Router).

SvenRoederer, 2019-12-08:
@captainkingcandy was you able to test the images?

captainkingcandy, 2019-12-09:
@SvenRoederer Yes, the image is running stable! I have to perform an reboot every two days, but this could be a configuration or log overrun problem :-) Thank you for your help Sven!

closed 2019-12-09 by captainkingcandy