freifunk-berlin/firmware #724: TP-Link CPE210 v2 fdfcd4a sysupgrade upstream-master (dev-daily) firmware bugs

everloop2, 2019-08-20:
https://lists.berlin.freifunk.net/pipermail/berlin/2019-August/039592.html

image [2019.08.19_upstream_master_freifunk-berlin-dev-daily-fdfcd4a-tplink_cpe210-v2-sysupgrade.bin](http://buildbot.berlin.freifunk.net/buildbot/unstable/ath79-generic/800/notunnel/freifunk-berlin-dev-daily-fdfcd4a-tplink_cpe210-v2-sysupgrade.bin):

-> flash via LuCI working
-> working: freifunk wizard + reboot
-> LuCI -> System -> Administration -> SSH - Access -> Save and Apply -> not working -> unsaved changes -> save and apply -> not working

wifi bugged:
-> wifi is on -> clients can not see AP -> disable: intern-ch13.cottbus.freifunk.net	-> same: clients can not see AP
-> wifi -> scan	-> shows only: intern-ch13.cottbus.freifunk.net (if activated) or nothing -> there are more wifis around

wifi bug hint: seen somebody mentioned in CPE210 bug threads different irqs

wifi not working: [2019.08.19_upstream_master_freifunk-berlin-dev-daily-fdfcd4a-tplink_cpe210-v2-sysupgrade.bin](http://buildbot.berlin.freifunk.net/buildbot/unstable/ath79-generic/800/notunnel/freifunk-berlin-dev-daily-fdfcd4a-tplink_cpe210-v2-sysupgrade.bin)
[   14.339957] ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
[   14.342144] ieee80211 phy0: Atheros AR9531 Rev:2 mem=0xb8100000, **irq=12**

wifi working: [2019.08.19_freifunk-berlin-development-1cfbb77-cpe210-v2-sysupgrade.bin](http://buildbot.berlin.freifunk.net/buildbot/unstable/ar71xx-generic/1541/default/freifunk-berlin-development-1cfbb77-cpe210-v2-sysupgrade.bin)
[   10.645916] ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
[   10.651507] ieee80211 phy0: Atheros AR9531 Rev:2 mem=0xb8100000, **irq=47**

niccokunzmann, 2019-09-17:
# CPE 210 v3

1. Ich hatte auch, dass das WLAN nicht geht. Nach längerem Warten ohne Strom geht es wieder.
  https://lists.berlin.freifunk.net/pipermail/berlin/2019-September/039760.html
2. Es scheint, als würde sich der Router immer wieder neu starten.
  https://monitor.freifunk-potsdam.de/grafana/d/000000008/node-overview?var-hostname=40-1-raguhner-schleife&refresh=5m&orgId=2
  [image]

everloop2, 2019-12-09:
closing

wifi has been fixed see: https://github.com/freifunk-berlin/firmware/issues/722#issuecomment-562153462 and https://github.com/freifunk-berlin/firmware/issues/722#issuecomment-562835293

other quirks needs testing again.

closed 2019-12-09 by everloop2