freifunk-berlin/firmware #725: Add support for Fritz Repeater 3000

db4rne, 2019-08-20:
It would be nice to have support in our firmware for the AVM Fritz!Repeater 3000. OpenWrt support is available: https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=148d29d47b1d555d0058dcc5dda450b74c4628ec

SvenRoederer, 2019-09-15:
An image was build for branch "upstream-master" I just cherry-picked for branch "upstream-1907".

@db4rne can you test the images?

SvenRoederer, 2019-10-01:
build fails for upstream-1907-branch (http://buildbot.berlin.freifunk.net/builders/ipq40xx-generic/builds/201)

SvenRoederer, 2019-10-04:
build is fixed for upstream-1907 also

db4rne, 2019-10-11:
Thanks for building the firmware. At the moment, there is no WAN defined when installing the firmware. I suspect that is because there is no WAN port in the vendor fw and there also is no switch in the hardware. Therefore, I'm suggesting to use LAN1 (eth1) as the designated WAN interface and LAN2 (eth0) as the designated LAN interface.

SvenRoederer, 2019-10-12:
I suspect OpenWrt upstream is not defining separate networks (https://git.openwrt.org/?p=openwrt/openwrt.git;a=blobdiff;f=target/linux/ipq40xx/base-files/etc/board.d/02_network;h=02cdd30b562203cec5472496bf0dc4869a38415c;hp=94fb26ce486982cc955925b86cff5de3bc73bdf7;hb=148d29d47b1d555d0058dcc5dda450b74c4628ec;hpb=5389db72e3265198986dfb50a831fb8f8264d346) as the LAN-ports are not color-coded (https://gzhls.at/i/98/01/1879801-n1.jpg). So it will be hard to distinguish them.
For our firmware we can patch this to two networks, but how do we document this? Have in mind that plugging in wrong order will bridge FF-Lan with users-private-LAN

SvenRoederer, 2019-11-21:
@db4rne isn't this a 3 wifi-card device?

What's the wizard saying in the WiFi-setup section?

db4rne, 2019-11-25:
Yes, that is a 3 card device. The Wizard isn't saying anything special while setting it up, but it configures the wrong card (the 5 GHz cards are divided into channel 36-64 and 100-165). So the wizard decides to configure the card that can only do 100-165 for channel 36, which won't come up obviously, but it is an easy error to fix for the user.

SvenRoederer, 2019-11-27:
@db4rne  can you create a bug-report for this, that we can track this?