freifunk-berlin/firmware #726: default iw package missing 'set plink_action'

goofus, 2019-08-29:
To ignore a 802.11s mesh point as described here, the 'set plink_action' setting has to be available in 'iw'. However, the default version doesn't support that: [iw.txt](https://github.com/freifunk-berlin/firmware/files/3555882/iw.txt)

Please include the 'iw-full' package as the default.
[iw-full.txt](https://github.com/freifunk-berlin/firmware/files/3555888/iw-full.txt)

SvenRoederer, 2019-08-29:
> as described here

What's "here"?

goofus, 2019-08-29:
> > as described here
> 
> What's "here"?

Sorry about that, forgot to add the [actual link](https://github.com/freifunk-gluon/gluon/issues/421)

SvenRoederer, 2019-09-01:
Excluding node from the mesh seems to be an uncommon / unwated intention / border-case, as also mentioned by the Gluon-guys. So changing the package-list will not be done in 1st place. 
But you can install the iw-full packkage directly from the openwrt-build-server and test.

goofus, 2019-09-02:
Ok, that's actually surprising for me. I thought meshing via LAN is common. Now, if two routers see each over the air shouldn't they stop meshing over wifi to save airtime?

goofus, 2019-09-03:
> But you can install the iw-full packkage directly from the openwrt-build-server and test.

root@router:~# opkg install iw-full
Installing iw-full (4.14-1) to root...
Downloading http://downloads.openwrt.org/releases/18.06-SNAPSHOT/packages/mips_24kc/base/iw-full_4.14-1_mips_24kc.ipk
Collected errors:
 * check_data_file_clashes: Package iw-full wants to install file /usr/sbin/iw
	But that file is already provided by package  * iw
 * opkg_install_cmd: Cannot install package iw-full.

mweinelt, 2019-12-26:
As far as I remember we considered this in Gluon, but the plink_action is not persisted across reconnects of mesh links.

nickbash11, 2020-07-08:
I did it [that way](https://github.com/nickbash11/mesh-blocker)

SvenRoederer, 2020-07-08:
@mweinelt Any idea if hostapd is working on this?

@nickbash11 your package makes it quite nice to use this function, even the bash script might not be the most ressource effective solution. I see some use for it and this package might provide a solution for the scenario mentioned by @goofus in comment https://github.com/freifunk-berlin/firmware/issues/726#issuecomment-527062302.

nickbash11, 2020-07-09:
@SvenRoederer I use it to prevent uninvited guests and any pests, because authsae decreases common speed of mesh network as I wrote in [this post](https://forum.openwrt.org/t/setting-up-authenticated-mesh-with-wpad-mesh/12399/98?u=nybash). It looks like an incompatibility with ath9k I think.

mweinelt, 2020-07-09:
> @mweinelt Any idea if hostapd is working on this?

Not to my knowledge.

PolynomialDivision, 2020-07-09:
@goofus u can use my fork https://github.com/Freifunk-Spalter/packages and build your own freifunk image with full iw package using openwrt image builder.
If u have any questions how to use it, just mention me in matrix channel.

SvenRoederer, 2020-07-09:
> Ok, that's actually surprising for me. I thought meshing via LAN is common. Now, if two routers see each over the air shouldn't they stop meshing over wifi to save airtime?
> In my opinion 802.11s is a central technology and hence all settings should be adjustable by default (as it was in latest stable).

By theory when meshing over LAN and meshing over WiFi, the Routing-daemon should prefer the wired link. That's what I would expect. So on the wireless link there should only be "hello" traffic, which should not have a huge impact. But in case the LAN-link fails, the Wifi-link should jump in quite quickly.
Also as @mweinelt  mentioned excluding a client will need need to be done on every reconnect.

@PolynomialDivision not sure how your mixup-feed will help with the initial issue of @goofus . I suggest to have a freifunk-berlin image with iw-full embedded to:

* clone the freifunk repo
* download freifunk-berlin-imagebuilder
* unpack the imagebuilder
* copy the iw-full*.ipk into the packages folder
* tar.xz the folder again
* modify an existing .package file in the packagelists folder to include iw-full in place of iw
* run the Makefile to use a preexisting imagebuilder (https://github.com/freifunk-berlin/firmware#customizing-make)

This should build an image with with iw-full in just some minutes. Alternatively you can just modify a .package file and recompile all yourself.

PolynomialDivision, 2020-07-09:
> @PolynomialDivision not sure how your mixup-feed will help with the initial issue of @goofus .

It is more a tidied-up feed. ;) That is exactly the usecase why I forked, since I wanted to build my own images with imagebuilder and easily add packages, compile stuff, or install things with opkg.

SvenRoederer, 2020-07-09:
> @SvenRoederer I use it to prevent uninvited guests and any pests, because authsae decreases common speed of mesh network as I wrote in [this post](https://forum.openwrt.org/t/setting-up-authenticated-mesh-with-wpad-mesh/12399/98?u=nybash). It looks like an incompatibility with ath9k (HT40 mode) I think.

As you talk on authsae: have a look at https://github.com/freifunk-gluon/gluon/issues/1636#issuecomment-464229021 which references that authsae was dropped upstream, as it was not working well.

nickbash11, 2020-07-10:
Anyway, devices with ath9k do not work properly with SAE.

SvenRoederer, 2020-07-10:
Not sure if we should capture this issue to discuss on SAE, esp for ath9k.

@goofus Not sure if my suggestion in https://github.com/freifunk-berlin/firmware/issues/726#issuecomment-656349226 will fit your needs and I don't see a great interest in switching to iw-full. But feel free to open a PR, I don't see a reason for vetoing.