freifunk-berlin/firmware #731: daily/upstream-master: unclear package lists

kls0e, 2019-10-09:
While making an effort to compile a 4MB image for some old 4/32 device, it is unclear to me which package lists should be used and which ones are relics from earlier commits. As of Oct 9th 2019, the content of /packages in upstream-master is 

`backbone_4MB.sh   
notunnel._4MB.txt              
tunnel-berlin-openvpn.txt
backbone_4MB.txt  
notunnel.txt                   
tunnel-berlin-tunneldigger_4MB.sh
backbone.txt     
tunnel-berlin-openvpn_4MB.sh   
tunnel-berlin-tunneldigger_4MB.txt
notunnel._4MB.sh  
tunnel-berlin-openvpn_4MB.txt  
tunnel-berlin-tunneldigger.txt`

If we could consolidate and elaborate on the package lists, I would be grateful. 

In particular:

- remove openvpn package lists?
- remove the dot in `notunnel.4_MB`?
- compared to current daily/upstream-1907, what is the difference between `mesh / default / notunnel` and what is each list meant to be used for?

SvenRoederer, 2019-10-12:
> notunnel._4MB.sh

this seems to bee a local file of you only. Please check with `git status`

> mesh 

* https://github.com/freifunk-berlin/firmware/commits/daily/upstream-1907/packages/mesh.txt
* https://github.com/freifunk-berlin/firmware/commit/c8f08c8a3a2946801bef505948ecf7d292b54151

> default / notunnel

* https://github.com/freifunk-berlin/firmware/pull/713

> remove openvpn package lists

The plan was to keep this packages till the openvpn-gateways are turned off, to give individual users the chance to build their own images when they cant migrate to tunneldigger yet.

kls0e, 2019-10-16:
thanks for constructively dissecting my whining.

SvenRoederer, 2019-10-16:
Sadly I don't know a proper place to document such historic knowledge.

closed 2019-10-16 by SvenRoederer