freifunk-berlin/firmware #735: wizard shows "Kein Passwort gesetzt!" dialog on fresh install

SvenRoederer, 2019-11-21:
### Bug report

**When setting up a new Node, the wizard displays the "Kein Passwort gesetzt!" dialog from OpenWrt/LuCI**

With our own fork "luci-mod-freifunk-ui" we added some codechanges to not display this message in the wizard. This seems to get broken by switching to LuCI-branch openWrt-1907.
**What is the expected behaviour?**

**Firmware Version:** git branch "master", revision 13f9c41

SvenRoederer, 2019-12-15:
[image]

znrR, 2020-03-26:
@sarumpaet will get asked by kaya if he can help with his knowhow on the old hedy solution

SvenRoederer, 2020-03-29:
I just had a test with rev 56c9a9d (notunnel-flavor) and did not see this problem anymore
* accessing 192.168.42.1 showed up the normal "welcome screen" of the wizard (no yellow textbox of OpenWrt-LUcI)
* on the following pages there was also no "yellow no password setup" dialog of OpenWrt

znrR, 2020-04-21:
needs testing - to do @Akira25