freifunk-berlin/firmware #736: Request Firmware support for Netgear N600 (WNDR3700v2)

xa-y-ax, 2019-11-25:
can we support the Netgear N600 (WNDR3700v2)?
https://openwrt.org/toh/netgear/wndr3700

SvenRoederer, 2019-11-25:
As I can see this router-family is supported :
* ath79: add WNDR3700 and WNDR3700v2 (https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=4dc7cb792a8e34b6ad97c7b8058c70311a06bd38)
* ramips: add factory image for WNDR3700v5 (https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=2fdd02cc7c3516c2cce2ace2c7b539e0821ecbe8)

This support seems also avail on OpenWrt-19.07 branch, so let's try ...

SvenRoederer, 2019-11-25:
btw. you can use the current imagebuilder to generate your own firmwareimages without running the compiler, see https://github.com/freifunk-berlin/firmware#customizing-make

pmelange, 2019-11-26:
There is a pull request to add this to the Hedy 1.0.X branch.  https://github.com/freifunk-berlin/firmware/pull/737 and is currently being built on buildbot http://buildbot.berlin.freifunk.net/#/builders/4/builds/18

After the build is finished and you test that it works, can you please leave a comment on PR #737? 

When the master branch builds again, another PR can be easily put together.

xa-y-ax, 2019-11-28:
Unfortunately the imagebuilder throws some errors - shoud i close this issue and switch to PR #737 ?

SvenRoederer, 2019-11-28:
do you have the error-message on hand?

xa-y-ax, 2019-11-28:
yep - but you already replied in PR #737 ...

Collected errors:
    opkg_install_cmd: Cannot install package kmod-leds-wndr3700-usb.
    Makefile:150: recipe for target 'package_install' failed
    make[3]: *** [package_install] Error 255
    Makefile:110: recipe for target '_call_image' failed
    make[2]: *** [_call_image] Error 2
    Makefile:206: recipe for target 'image' failed
    make[1]: *** [image] Error 2
    make[1]: Leaving directory
    '/home/xayax/firmware/imgNT0enz/freifunk-berlin-imagebuilder-1.0.4-ar71xx-generic.Linux-x86_64'
    Makefile:204: recipe for target '.stamp-images' failed
    make: *** [.stamp-images] Error 2

SvenRoederer, 2019-11-28:
this PR ist around master-branch, PR737 is about backporting to Hedy-1.0.x

the error shown says you are using the Hedy-1.0.4 imagebuilder. have you tried with imagebuilder from master?

pmelange, 2019-11-28:
There currently isn't a master branch.  https://util.berlin.freifunk.net/hardware?name=tl-wdr3600-v1,tplink_tl-wdr3600&complete=true

As stated https://github.com/freifunk-berlin/firmware/issues/736#issuecomment-558689981 "When the master branch builds again, another PR can be easily put together."

SvenRoederer, 2019-11-29:
@pmelange keeping only the last 10 builds is sometimes a bit to less, that's why Issue #385 would be nice.

Master-builds are broken as seen on issue #734. in branch "master_gluon-patch" I'm working on a solution, to fix this problem of out Makefile.

SvenRoederer, 2019-11-29:
@xa-y-ax  I did the following to get a working image-set for WNDR3700-v2
> git checkout master
> wget http://firmware.berlin.freifunk.net/unstable/ath79-generic/17/freifunk-berlin-imagebuilder-Development-ath79-generic.Linux-x86_64.tar.xz
> echo >> profile/ath79-generic.profiles "netgear_wndr3700v2"
> make IB_FILE=freifunk-berlin-imagebuilder-Development-ath79-generic.Linux-x86_64.tar.xz TARGET=ath79-generic images

xa-y-ax, 2019-11-30:
hm. this also doesn't works for me.
`make images IB_FILE=freifunk-berlin-imagebuilder-Development-ath79-generic.Linux-x86_64.tar.xz PACKAGE_LIST_DEFAULT=tunnel_berlin_tunneldigger
IB_FILE explicitly defined; using it for building firmware-images
mkdir -p /home/xayax/firmware/firmwares/ar71xx-generic
./assemble_firmware.sh -p "wndr3700v2" -i freifunk-berlin-imagebuilder-Development-ath79-generic.Linux-x86_64.tar.xz -e /home/xayax/firmware/embedded-files -t /home/xayax/firmware/firmwares/ar71xx-generic -u "default tunnel-berlin backbone"
/home/xayax/firmware/firmwares/ar71xx-generic
Extract image builder freifunk-berlin-imagebuilder-Development-ath79-generic.Linux-x86_64.tar.xz
Building a profile for wndr3700v2
Building usecase default
Using package list default
embedding files from /home/xayax/firmware/embedded-files.
make[1]: Entering directory '/home/xayax/firmware/imgGAGTzp/freifunk-berlin-imagebuilder-Development-ath79-generic.Linux-x86_64'
make[2]: Entering directory '/home/xayax/firmware/imgGAGTzp/freifunk-berlin-imagebuilder-Development-ath79-generic.Linux-x86_64'
Profile "wndr3700v2" does not exist!
Use "make info" to get a list of available profile names.
Makefile:192: recipe for target '_check_profile' failed
make[2]: *** [_check_profile] Error 1
make[2]: Leaving directory '/home/xayax/firmware/imgGAGTzp/freifunk-berlin-imagebuilder-Development-ath79-generic.Linux-x86_64'
Makefile:199: recipe for target 'image' failed
make[1]: *** [image] Error 2
make[1]: Leaving directory '/home/xayax/firmware/imgGAGTzp/freifunk-berlin-imagebuilder-Development-ath79-generic.Linux-x86_64'
Makefile:204: recipe for target '.stamp-images' failed
make: *** [.stamp-images] Error 2`

any hints for me? I will now try Stefans build:  http://lernekultur.de/ffbf/

SvenRoederer, 2019-11-30:
Seems you did not use die Imagebilder of the URL I noted for wget. On recent builds I backported the renaming of "wndr3700v2" to "netgear_wndr3700-v2", to ease upgrades to further OpenWRT releases.

Just retry with this changed profile-name.
You can also unpack the imagebuilder an run "make info" to get a list of avail profiles.

SvenRoederer, 2019-12-08:
I rebased branch ```issue736_add-WNDR3700``` where the image should be build as ```netgear_wndr3700-v2```

SvenRoederer, 2019-12-08:
> make images IB_FILE=freifunk-berlin-imagebuilder-Development-ath79-generic.Linux-x86_64.tar.xz PACKAGE_LIST_DEFAULT=tunnel_berlin_tunneldigger
> IB_FILE explicitly defined; using it for building firmware-images
> mkdir -p /home/xayax/firmware/firmwares/ar71xx-generic

@xa-y-ax Just saw your mistake: you missed the ```TARGET=ath79-generic``` definition on your make call. This let the make buildinf for the default ar71xx-generic, whereas you supplied an imagebuilder for ath79-generic.

SvenRoederer, 2020-02-14:
was anyone able to test PR #741 and confirm?

SvenRoederer, 2020-09-30:
> was anyone able to test PR #741 and confirm?

As nobody seems to have any interest in testing the fix since 10 months, I'll close soon.

closed 2020-10-15 by SvenRoederer