freifunk-berlin/firmware #746: close all issues relating to Freifunk-Wizard-2

SvenRoederer, 2019-12-17:
As per comment https://github.com/freifunk-berlin/firmware/issues/626#issuecomment-553527201 of @andibraeu there seems nobody working on the new wizard anymore. The last relevant commits to the frontend or backend are several years ago.
So let's close the issues relating to this work.

* https://github.com/freifunk-berlin/firmware/issues/482
* https://github.com/freifunk-berlin/firmware/issues/481
* https://github.com/freifunk-berlin/firmware/issues/397
* https://github.com/freifunk-berlin/firmware/issues/396
* https://github.com/freifunk-berlin/firmware/issues/395
* https://github.com/freifunk-berlin/firmware/issues/394
* https://github.com/freifunk-berlin/firmware/issues/393
* https://github.com/freifunk-berlin/firmware/issues/392

At https://raw.githubusercontent.com/freifunk-berlin/firmware/d92a2c0f48e9a366c510720443f1c3ff540bf0ba/packages/default.txt is a [packages.list](https://github.com/freifunk-berlin/firmware/files/3975791/wizard2-packagelist.txt) that should create a usable image. Last commit of the branch was for such images was d92a2c0f48e9a366c510720443f1c3ff540bf0ba.

@sarumpaet @pmelange @kls0e with the above references you should be able to restart working on this project, if you like.

SvenRoederer, 2019-12-17:
also to mention @andrenarchy

SvenRoederer, 2019-12-29:
Also the relating [Milestone](https://github.com/freifunk-berlin/firmware/milestone/6) should be closed.

SvenRoederer, 2020-04-27:
We should also think of removing the code from the repo, when it's not maintained anymore as it seems to cause confusion (https://github.com/freifunk-berlin/firmware-packages/pull/165)

Akira25, 2020-04-27:
I agree on the idea to close all issues related to wizard2. But on the other hand, I would like so save this code in any way, in case we decide to continue on that project. Maybe we could push it to another repo of freifunk-berlin?

SvenRoederer, 2020-04-27:
you can always go back in history and restore it by reverting the commit which deletes the code.

xa-y-ax, 2020-05-05:
as discussed on 5.5.2020 we close the issues relating to Freifunk-Wizzard-2

SvenRoederer, 2020-05-06:
wizard2-code was removed from packages-repo and a tag "remove_wizard2" was added for reference.

SvenRoederer, 2020-05-06:
all relating issues should have been closed now.

closed 2020-05-06 by SvenRoederer