freifunk-berlin/firmware #747: change versioning scheme to based on release date

SvenRoederer, 2019-12-18:
Some points should make us think of changing the current versioning-scheme.
* upstream removal of "CONFIG-Option VERSION_NICK" (#515) which makes it hard to find a proper place to store our Releasename (Kathleen, Hedy, Gonzo). The way choosen for "Gonzo-rc"-builds was to combine the numeric version with the releasename, which instantly caused such issues --> #710 
* Semantic versioning: on the [Mailinglist there was some discussion](https://lists.berlin.freifunk.net/pipermail/berlin/2019-June/039493.html) if it is really handy for enduser-software. Modifying this scheme, from what was agreed in #145, and dropping the releasename will avoid confusion as we have seen when going from Kathleen to Hedy. Even I support the idea of honoring famous (female) persons of the IT-sector.

I vote for going to something like "Freifunk Berlin 2020.02.0" for a release scheduled for early 2020. Consequently the 1st maintenance-release will become "Freifunk Berlin 2020.02.1" and a prerelease will be "Freifunk Berlin 2020.02.0-rc1". This will not strictly meet the SemVer-definition, as we will change the major-number on a annual basis, even if there are no incompatible API changes.

everloop2, 2019-12-19:
numbers: appreciate it. Its far more easy to compare.

znrR, 2020-03-26:
Pro Nickname: "Marketing" because we can dedicate a major release to an "influentual" person such as Hedy Lamar
Con Nickname: The field to store that nickname has been removed by OWRT and it would mean we need to attach it to some other field (eg. vendor) which requires follow up work on various currently unknown non firmware related projects to hide that information again.

znrR, 2020-04-21:
proposal: could also be placed in the footer and no longer be part of the "real" version number 
[image]

znrR, 2020-04-21:
technically this would look like:
$CONFIG_VERSION_DIST Firmware Release $CONFIG_VERSION_NICK! $CONFIG_VERSION_DEDICATION

znrR, 2020-04-21:
agreed as to do for @SvenRoederer

znrR, 2020-04-21:
also we agreed on changing the versioning from a consecutive 1.1.XX to timestamp based 2020.XX.Y which is similar to the OWRT scheme.