freifunk-berlin/firmware #748: compiling fails on missing dependency luci-lib-httpclient

kls0e, 2019-12-19:
I'm having a hard time compiling both upstream-master and Hedy-1.0.x

Both fail with `Collected errors:

 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-owm-cmd:
 *      luci-lib-httpclient
 * opkg_install_cmd: Cannot install package luci-app-owm-cmd.`

Build environment should be okay.

SvenRoederer, 2019-12-19:
For Hedy-1.0.x the build was passing this morning on the buildbots and the package was created (http://firmware.berlin.freifunk.net/unstable/ar71xx-generic/83/packages/packages/mips_24kc/luci/luci-lib-httpclient_git-19.271.72080-7b230b0-1_all.ipk)

There was no recent change, so I don't expect it relates to a specific version. Please try: ```rm -rf openwrt/build-dir/target*``` and start make again.
If this doesn't help run ```make clean``` but this will cause rebuilding of openwrt-hosttools and openwrt-toolchain ...

kls0e, 2019-12-20:
thank you Sven, i still get 
```
WARNING: Makefile 'package/feeds/routing/luci-app-bmx6/Makefile' has a dependency on 'luci-lib-httpclient', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-dnscrypt-proxy/Makefile' has a dependency on 'luci-lib-httpprotoutils', which does not exist
WARNING: Makefile 'package/feeds/packages_berlin/luci-app-owm/Makefile' has a dependency on 'luci-lib-httpclient', which does not exist
```

SvenRoederer, 2019-12-21:
I assume you see this problem also for the "daily/upstream-1701" branch. The [last build of the branch](http://buildbot.berlin.freifunk.net/#/changes/226) passed on all buildslaves for all enabled targets.
Can you checkout a copy of this commit and check with ```git status``` that all files are up-to-date?

kls0e, 2019-12-24:
thanks for your support :)
had it on daily/upstream-master and on Hedy-1.0.x 
I have sorted it out by correcting my ./configs, pardon.

closed 2019-12-24 by kls0e

SvenRoederer, 2019-12-24:
Great to hear. Looking forward to your next contributions.