freifunk-berlin/firmware #752: Speedport W504V: Kernel panic / Wifi modes

kls0e, 2019-12-31:
Hi, 

I have compiled Hedy-1.0.5alpha (thanks Sven!) for [the mentioned lantiq-xway danube based target](https://openwrt.org/toh/t-com/spw504v) and after finishing the freifunk wizard, the board panics as follows: 
```
[   52.794602] Process hotplug-call (pid: 1479, threadinfo=82932000, task=8390de90, tls=77f06d48)
[   52.802951] Stack : 803efbbc 80410000 2c8fe372 82ed9de0 82c70bd0 82ed9d50 80410000 82ed9000
          838ace60 82ed9400 8380db0c 00000000 00000000 00000000 0000006c 00000004
          00500003 00000000 86dd3a00 00000000 00000000 00000000 00000000 00000000
          00000000 00000000 ff020000 00000000 00000000 00000016 00000014 9a98597f
          82c70bd0 82ed9d50 0000006c 82daf760 838ace40 82c70b40 838ace40 0000006c
          ...
[   52.837418] Call Trace:[<82daf760>] 0x82daf760 [mac80211@82d80000+0x6a2e0]
[   52.844164] [<82dafac4>] 0x82dafac4 [mac80211@82d80000+0x6a2e0]
[   52.849893] [<82dac848>] 0x82dac848 [mac80211@82d80000+0x6a2e0]
[   52.855641] [<82db17c8>] 0x82db17c8 [mac80211@82d80000+0x6a2e0]
[   52.861415] [<82db2ae0>] 0x82db2ae0 [mac80211@82d80000+0x6a2e0]
[   52.867134] [<83341fc0>] 0x83341fc0 [nf_conntrack@83340000+0xc240]
[   52.873099] [<83342104>] 0x83342104 [nf_conntrack@83340000+0xc240]
[   52.879076] [<802c9508>] 0x802c9508
[   52.882520] [<82db2e80>] 0x82db2e80 [mac80211@82d80000+0x6a2e0]
[   52.888199] [<800b2c9c>] 0x800b2c9c
[   52.891596] [<8029b148>] 0x8029b148
[   52.894977] [<8029b264>] 0x8029b264
[   52.898356] [<8029b78c>] 0x8029b78c
[   52.901735] [<802a76f0>] 0x802a76f0
[   52.905097] [<8032ef88>] 0x8032ef88
[   52.908475] [<802c9c04>] 0x802c9c04
[   52.911856] [<80332164>] 0x80332164
[   52.915232] [<802c9c04>] 0x802c9c04
[   52.918632] [<80331f98>] 0x80331f98
[   52.921991] [<803558f8>] 0x803558f8
[   52.925369] [<80357a1c>] 0x80357a1c
[   52.928770] [<802d412c>] 0x802d412c
[   52.932133] [<80357a1c>] 0x80357a1c
[   52.935507] [<80357a54>] 0x80357a54
[   52.938886] [<8005dba4>] 0x8005dba4
[   52.942267] [<803577a8>] 0x803577a8
[   52.945646] [<8005c700>] 0x8005c700
[   52.949026] [<8005c970>] 0x8005c970
[   52.952403] [<80055a58>] 0x80055a58
[   52.955786] [<8002d010>] 0x8002d010
[   52.959161] [<800588fc>] 0x800588fc
[   52.962545] [<8000d780>] 0x8000d780
[   52.965920] [<80010000>] 0x80010000
[   52.969300] [<800027f8>] 0x800027f8
[   52.972671]
[   52.974107]
Code: afb40010  afb3000c  afb20008 <8ccc000c> 3c117974  3c036e64  24636f6d  26316573  8cd70000
[   52.983941] ---[ end trace 62b59bda3b9cadf5 ]---
[   52.993115] Kernel panic - not syncing: Fatal exception in interrupt
[   53.000516] Rebooting in 3 seconds..
```
iw list says 
```
        valid interface combinations:
                 * #{ managed } <= 2048, #{ AP, mesh point } <= 8, #{ P2P-client, P2P-GO } <= 1, #{ IBSS } <= 1,
                   total <= 2048, #channels <= 1, STA/AP BI must match, radar detect widths: { 20 MHz (no HT), 20 MHz, 40 MHz }
```
Does that mean that 802.11s+AP is fine but IBSS+AP is not? Would this cause a kernel panic?

SvenRoederer, 2020-01-01:
> iw list says
>
> ```
>         valid interface combinations:
>                  * #{ managed } <= 2048, #{ AP, mesh point } <= 8, #{ P2P-client, P2P-GO } <= 1, #{ IBSS } <= 1,
>                    total <= [...]

It's not the 1st board that it not supporting AdHoc anymore. Please check the issues tagged mit "mesh (11s) - IBSS" - flag:
* https://github.com/freifunk-berlin/firmware/labels/mesh%20%2811s%29%20-%20IBSS

kls0e, 2020-01-01:
thank you, I wanted to make sure I do interpret the output in the right way.

SvenRoederer, 2020-01-01:
Regarding the kernel-panic: There is a chance that this relates to the attempt to setup a STA-AdHoc combination.
Can you try with our master-build (802.11s in the wizard, more recent kernel)?

SvenRoederer, 2020-01-01:
btw: "Hedy-1.0.5alpha" are you using the regular "Hedy-1.0.x" branch (66ce1f6c33716846a7657561cb90e6712d424b00) or the branched-off version with kernel 4.4.205 (90b63aa2de4a4bc42b2a8583e26b8c5c144992ab)? When using kernel-4.4.205 there might also be a relation to the crashes @torte71 mentioned in https://github.com/freifunk-berlin/firmware/pull/749

everloop2, 2020-01-02:
Did build working firmware some time ago -> 802.11s + AP working and stable

Speedport is only able to do 2.4 **or** 5Ghz at same time -> "fix": add 2.4 or 5Ghz USB dongle ;)

firmware: https://wiki.freifunk-cottbus.de/firmware:downloads

Speedport W504V Typ A minimal Freifunk Berlin (Dev-SAm0815 rev dd2f75e), 16.07.2019

OpenWrt Wiki: https://openwrt.org/toh/t-com/spw504v

Brnboot replace -> Easy & short guide (serial + LAN) -> https://openwrt.org/toh/arcadyan/arv7506#installation

kls0e, 2020-01-02:
excellent. thank you!

kls0e, 2020-01-02:
> 
> 
> btw: "Hedy-1.0.5alpha" are you using the regular "Hedy-1.0.x" branch ([66ce1f6](https://github.com/freifunk-berlin/firmware/commit/66ce1f6c33716846a7657561cb90e6712d424b00)) or the branched-off version with kernel 4.4.205 ([90b63aa](https://github.com/freifunk-berlin/firmware/commit/90b63aa2de4a4bc42b2a8583e26b8c5c144992ab))? When using kernel-4.4.205 there might also be a relation to the crashes @torte71 mentioned in #749

I am using the rebranched kernel-4.4.205 but this seems to be unrelated.

closed 2020-01-02 by kls0e

SvenRoederer, 2020-01-02:
> Did build working firmware some time ago -> 802.11s + AP working and stable

so the crashes seen by @kls0e are very likely related to the olst 4.4-kernel series of Hedy-1.0 (Lede-17.01).

SvenRoederer, 2020-01-02:
> Speedport is only able to do 2.4 **or** 5Ghz at same time -> "fix": add 2.4 or 5Ghz USB dongle ;)

@everloop2 did the wizard gave you the correct option-list to select the WLan-frequency?

everloop2, 2020-01-03:
at ffwizard it shows:

```
5 Ghz Wifi (RADIO0)

Mesh-IP: xxxx

Mesh Mode: 802.11s OR Ad-HOC (veraltet)

Freifunk-SSID: cottbus.freifunk.net

DHCP-Network: xxxx
```
-> can not change to 2.4 GHz at ffwizard -> did set frequency after save and first reboot in `LuCI -> Wireless`

-> did see: if there is a second wifi (USB dongle), its recognized and a 2nd IP for mesh can be set ;)

everloop2, 2020-01-03:
> > Did build working firmware some time ago -> 802.11s + AP working and stable
> 
> so the crashes seen by @kls0e are very likely related to the olst 4.4-kernel series of Hedy-1.0 (Lede-17.01).

dunno -> think started at least with 4.9 / 4.14 kernel (Sam_experimental builds), 4.19 works flawless and "fast" ;)