freifunk-berlin/firmware #754: upstream-1907: Freifunk-Wizard generates invalid ipv4 netmask for interface dhcp

kls0e, 2020-01-03:
Hi,

there seems to be a bug in the current upstream-1907 wizard config generation scripts that results in an invalid netmask entry for the ipv4 dhcp network configuration. The device is not reachable via IPV4 after the wizard is completed. A sample screenshot by another freifunker is attached, there seems to be a misplaced `%tunl0` or `%tun10` in the netmask field value (iirc using a tunneldigger image).
[image]

SvenRoederer, 2020-01-07:
On the weekend I also had problems setting up a node from scratch with "SAm0815-experimental_master (ef318a5). Later I was able to trigger this also on an older build of "1907-daily (863ec85). 
Both Nodes also used tunneldigger-uplink. On the 1907-daily (863ec85) I had a netmask of '255.255.255.240%ifb0', on the SAm0815-experimental_master I don't know.

* What was the exact version you used?
* Have you seen this on Tunneldigger-uplinks only?
* Have you seen this on master builds also?
* Any idea when it broke?

SvenRoederer, 2020-01-09:
Did a short test: 

* good: 013a01b0c5bb34b4c46e1839f70cd07787379377
* broken: 4de864fbb5e66065b7a4a6af3a9fccc75f17b452

So it's very likely related to some upstream change ... I don't expect 97a920482ac4fc0818dc3161ba51a250928edc2e to cause this problem

kls0e, 2020-01-09:
Thanks for looking into this.

kls0e, 2020-01-09:
>     * What was the exact version you used?
> 
>     * Have you seen this on Tunneldigger-uplinks only?
> 
>     * Have you seen this on master builds also?
> 
>     * Any idea when it broke?

For the time being I cannot give any precise answers to these questions, pardon.

SvenRoederer, 2020-01-09:
Let me add; I was able to reproduce when selecting 
* join Freifunk-net (do not share uplink)
* using the notunnel image

SvenRoederer, 2020-01-11:
I checked around between the changes and found that up to b86d6672a639f24f15820298b1ce82de31851843 all is fine, since commit 9795c9b98a150f71935d4b7809e044727b2b2231the failure can be seen. So it's something the comes from the LuCI-repo. In the list of upstream-changes I don't see one, which I would guess is causing the fault in the wizard.

@kls0e I replaced the original commit, covering all changes by some smaller commits from the daily/upstream branch. Can you ```git bisect``` this?

SvenRoederer, 2020-02-02:
I had a closer look on upstream changes an found https://github.com/openwrt/luci/commit/1d2b4c777fd47f8d520b21d7085d3e7eb387cf02 to cause the problem. Applying a patch to revert this (https://github.com/SvenRoederer/freifunk-berlin-firmware/blob/issue754/patches/packages/luci/0001-Revert-luci-lib-ip-support-scoped-IPv6-addresses.patch) fixes this problem

SvenRoederer, 2020-02-05:
Applying the mentioned patch (SvenRoederer@6840e68) on master (b1e0f151f1b1f44b27b02cf5ca464a90d4738b34) fails, which seems to be related https://github.com/openwrt/luci/commit/b349428e167c5c129ce352aa6d391fea85122fd4.
Just doing a test build, but I expect from the commit message to fix the problem causing this issue.

SvenRoederer, 2020-02-05:
My guessing was wrong, the code still fails. As of this I made https://github.com/freifunk-berlin/firmware-packages/commit/4067c3461d4f909574e6efaff76684dbeff31e43 which adds some debug to the wizard 

>    tools.logger("dhcpmeshnet.addr: " .. dhcpmeshnet:minhost():string())
    tools.logger("dhcpmeshnet.mask: " .. dhcpmeshnet:mask():string())
    tools.logger("prenetconf.mask: " .. prenetconfig.netmask)

This show i the syslog:

> ffwizard: dhcpmeshnet.mask: 255.255.255.240
> ffwizard: prenetconf.mask: 255.255.255.240%ifb1

so for some reason assigning ```dhcpmeshnet:mask():string()``` to ```prenetconfig.netmask``` causes adding of an interfacename (https://github.com/freifunk-berlin/firmware-packages/blob/1226ad42fe81eba6bd74df786320976cc154f006/utils/luci-app-ffwizard-berlin/luasrc/model/cbi/freifunk/assistent/wireless.lua#L246).

SvenRoederer, 2020-02-06:
filled a bug on luci-upstream (https://github.com/openwrt/luci/issues/3609)

SvenRoederer, 2020-02-07:
Tests have shown that https://github.com/freifunk-berlin/firmware-packages/commit/ea5666c516fa458bd94e7c1a8d79ed17c6c5d105 might fix this.

SvenRoederer, 2020-02-07:
Even the problem persists in upstream-code changing the order of assignments fixes this problem in our wizard (https://github.com/freifunk-berlin/firmware-packages/commit/b03e01ed5a3b691a19cf284627aab9820ce6c2b8).

closed 2020-02-07 by SvenRoederer

SvenRoederer, 2020-02-14:
Upstream was able to fix the cause with https://github.com/openwrt/luci/commit/c2eeb09bd99b03eead837d8dbfb3f0d8046ccc65. As the order of assignments is not changing the functionality of our code, I'll keep it like it is.