freifunk-berlin/firmware #755: auto-install bbbdigger after remote sysupgrade

kls0e, 2020-01-09:
Hi, 
in case I am upgrading the firmware of a remote tunneldigger node via ssh, a manually installed bbbdigger backbone tunnel will be lost and the router is no longer reachable from within freifunk berlin backbone. I know this has been pointed out before - I would appreciate it if we found a way of how to mitigate this issue.

Akira25, 2020-01-09:
Hi kl0se,

maybe you could have a look at those two function from my script.

The first one generates a list of the currently installed packages and save them at a location, which is secured by sysupgrade over reflash.
[get_package_list()](https://github.com/Akira25/autoupdate/blob/53e93881241c6c0ecedfd479ee6dc4dedd6af539/files/libautoupdate.sh#L182)
[set_preserved_backup_dir()](https://github.com/Akira25/autoupdate/blob/53e93881241c6c0ecedfd479ee6dc4dedd6af539/files/libautoupdate.sh#L105)

After the upgrade a uci-defaults-script should handle, to install all missing packages.
[#get the list of packages and compare](https://github.com/Akira25/autoupdate/blob/53e93881241c6c0ecedfd479ee6dc4dedd6af539/files/config_defaults.sh#L26)

You could port those functions into a separate commit, as the finalization of my script may take some further time.

pmelange, 2020-01-10:
See #701.