freifunk-berlin/firmware #757: upstream-1907: build fails early

kls0e, 2020-01-12:
what has happened to the upstream-1907 target? as of today, building a vanilla cloned upstream-1907 fails very early at 
`Applying: ath79: add support for TP-Link CPE510-v2/v3
error: patch failed: target/linux/ath79/image/generic-tp-link.mk:146
error: target/linux/ath79/image/generic-tp-link.mk: patch does not apply
Patch failed at 0001 ath79: add support for TP-Link CPE510-v2/v3
`

SvenRoederer, 2020-01-13:
https://github.com/openwrt/openwrt/commits/openwrt-19.07/target/linux/ath79/image/generic-tp-link.mk shows that https://github.com/openwrt/openwrt/commit/19ff3f5105791a31b27932e1bf2ad9e150604b6e changed the file.
This change made applying the patch "ath79: add support for TP-Link CPE510-v2/v3" (https://github.com/freifunk-berlin/firmware/blob/daily/upstream-1907/patches/openwrt/0005-ath79-add-support-for-cpe510v2v3.patch) fail.
So our patch needs to be updated or removed.

SvenRoederer, 2020-01-13:
> what has happened to the upstream-1907 target? as of today, building a vanilla cloned upstream-1907 fails very early at

btw. it'S always very helpful to include a git rev. a rev will always point to a specific point in the repo. referencing to today will be different when reading the issue next week.

SvenRoederer, 2020-01-26:
fixed by 12ad1a9104b1fdfe1d7840ad5f8dd3e6eddceed0

closed 2020-01-26 by SvenRoederer