freifunk-berlin/firmware #758: upstream-1907: notunnel_4MB

kls0e, 2020-01-17:
Hi, what would be the contents for the notunnel_4MB.txt and notunnel_4MB.sh packages list files? I know its current content says that images would be too big for 4MB, but I'd still like to try building, especially with ath79-target and maybe removing some other stuff individually.

SvenRoederer, 2020-01-17:
Isn't ```git revert 5725069daae00b3cc301d70b1be4d9deebb5e8d2```doing the trick?
On console you can use ```git log packages/notunnel_4MB.txt``` to see the list of changes for a specific file. On github you can just click the History-button, in the upper right, then displaying the specific file.

SvenRoederer, 2020-01-17:
> especially with ath79-target

Before I changed the package-list I did a test-build with ath79-tiny. Regarding removing packages, have also a look into https://github.com/freifunk-berlin/firmware/issues/626

closed 2020-01-18 by kls0e