freifunk-berlin/firmware #759: add support for TP-Link MR3020 V3

kls0e, 2020-01-22:
Hi,
there is another usb-powered microuter we should support: TP-Link [MR3020 V3](https://openwrt.org/toh/tp-link/tl-mr3020_v3). V3 has 8/64 and is MediaTek MT7628NN-based, it is a current model that is being sold (as off 1/2020). It comes with only one eth-Port so for tunneldigger and notunnel, the config needs to be adjusted to make eth0 act as wan source.

SvenRoederer, 2020-01-25:
for the problem of singe-ETH-port devices we just need to finish the script ```sharenet-switch.sh```  mentioned in 
https://github.com/freifunk-berlin/firmware-packages/blob/5c1e3afd44321146d809fb984c74d620a1d4de46/addons/freifunk-berlin-configtools/ReadMe.txt#L10

SvenRoederer, 2020-10-22:
> for the problem of singe-ETH-port devices we just need to finish the script `sharenet-switch.sh` mentioned in
> https://github.com/freifunk-berlin/firmware-packages/blob/5c1e3afd44321146d809fb984c74d620a1d4de46/addons/freifunk-berlin-configtools/ReadMe.txt#L10

See #826 for tracking this

SvenRoederer, 2020-11-01:
Device-support added in 91bfb591d8381344aef5ce610e933751a19a491c; open issues are tracked in #826 and #292.

SvenRoederer, 2020-11-01:
just seen it's about the v3, not the v2