freifunk-berlin/firmware #760: freifunk-berlin-generic.mk uses wildcard command that's not present on every system

booo, 2020-01-23:
The absence of the wildcard command breaks the freifunk-berlin-generic.mk code.

If I try to execute the `freifunk-berlin-generic.mk` from the "TOPDIR" in a downloaded SDK I get this error:

```
➜  openwrt-sdk-19.07.0-x86-generic_gcc-7.5.0_musl.Linux-x86_64 make -f feeds/freifunk_berlin/freifunk-berlin-generic.mk
feeds/freifunk_berlin/freifunk-berlin-generic.mk:10: *** did not find luci.mk in any feed.  Stop.
```

This is most likely because I don't have `wildcard` installed on my system:

```
➜  ~ which wildcard
wildcard not found
```

`wildcard` comand is used in line 5 of the makefile. See https://github.com/freifunk-berlin/firmware-packages/blob/456e434dab9e1e63c859f8d1ac3c3d746c51cc8d/freifunk-berlin-generic.mk#L5

We should use a command that is present on all systems!

Searching for a luci.mk file in the other feeds seems like a strange idea anyway. Maybe we should just duplicate the code that is important to us and live with that. Or find a proper way to include this makefile (luci.mk) in our makefiles.

SvenRoederer, 2020-01-25:
The $(wildcard) function is part of GNU make (https://www.gnu.org/software/make/manual/html_node/Wildcard-Function.html).

booo, 2020-01-26:
Yes, I figured that out as well.

Maybe it's connected to the use of $(TOPDIR) instead. I have to check again.

SvenRoederer, 2020-01-26:
I just pushed 5876fb2c9207cbb801eec3b426c312b8bcab1409, which might be handy to configure the SDK. So its an easy way, to just download the SDK, install, modify package and run ```make package/<name>/compile```

SvenRoederer, 2020-10-23:
> Yes, I figured that out as well.
> 
> Maybe it's connected to the use of $(TOPDIR) instead. I have to check again.

@boo was you able to check again?

SvenRoederer, 2020-11-01:
>> The $(wildcard) function is part of GNU make

> Yes, I figured that out as well.

So the initial issue is fixed.

closed 2020-11-01 by SvenRoederer