freifunk-berlin/firmware #762: Eigenes Berliner Freifunk Image für Mikrotik RB 911 ohne WLan

SvenSFS, 2020-01-27:
Mir ist aufgefallen das eine eigenes Image auf Basis der 
Berliner Firmware zum einen kein Wirzard durchläuft.
Die Profile werden nur als Fragezeichen "?" angezeigt und
es gibt kein Funktionierendes W-Lan.

Es wird zwar das ath9k Kernel Modul geladen aber 
W-Lan weigert sich beharrlich zu Starten.

Das Board ist ein RouterBOARD 911G-5HPacD

SvenRoederer, 2020-01-28:
Auf welchem code basiert das Image (Commit)?
Was genau ist mit "eine eigenes Image auf Basis der Berliner Firmware" gemeint?

O.g. Punkte wären bei Nutzung des Bug-report templates (https://github.com/freifunk-berlin/firmware/issues/new?template=bug_report.md) abgefragt worden.

SvenRoederer, 2020-05-06:
@SvenSFS  Ist das noch relevant?

closed 2020-12-20 by SvenRoederer