freifunk-berlin/firmware #767: Archer-C7-V4: wireless.lua:143: attempt to concatenate field 'mode' (a nil value)

ikreb7, 2020-03-15:
Hello,

I want to install the upstream-1907 (119) vom 2020-03-15 06:29:17 version for the archer c7 with version 4.

If I go through the setup till the end I click ``Save and Reboot`` and get then this error message:

<code>
Failed to execute cbi dispatcher target for entry '/admin/freifunk/assistent/wireless'.
The called action terminated with an exception:
...b/lua/luci/model/cbi/freifunk/assistent/wireless.lua:143: attempt to concatenate field 'mode' (a nil value)
stack traceback:
	...b/lua/luci/model/cbi/freifunk/assistent/wireless.lua:143: in function 'i'
	/usr/lib/lua/luci/model/uci.lua:203: in function 'foreach'
	...b/lua/luci/model/cbi/freifunk/assistent/wireless.lua:102: in function 'write'
	...b/lua/luci/model/cbi/freifunk/assistent/wireless.lua:86: in function 'parse'
	/usr/lib/lua/luci/cbi.lua:201: in function 'parse'
	/usr/lib/lua/luci/cbi.lua:201: in function 'parse'
	/usr/lib/lua/luci/cbi.lua:559: in function 'parse'
	/usr/lib/lua/luci/dispatcher.lua:871: in function </usr/lib/lua/luci/dispatcher.lua:863>
</code>

If you need further information I can add them.

SvenRoederer, 2020-03-18:
> upstream-1907 (119)

What version is that? What's the git-rev?
Why not using the BUG-report template (https://github.com/freifunk-berlin/firmware/issues/new/choose)?

ikreb7, 2020-03-19:
> What's the git-rev?
git-19.249.30590-93abd43

SvenRoederer, 2020-03-22:
Looks like you provided the LuCI-revision, which doesn't help. The section floowing this is relevant for our firmware.

this is the line from one of my routers; see the section after the "/"
> Powered by LuCI branch (git-19.249.30590-0b5e6d7) / Freifunk Berlin Dev-daily-1907 009f2ea

ikreb7, 2020-03-23:
Sorry.

cat /etc/openwrt_version
ad9b28b

SvenRoederer, 2020-03-24:
I just made a debug-build for this issue (56c9a9d). Can you retested with the following image:

> sysupgrade -n http://77.245.36.54:8000/firmwares/unstable/ath79-generic/notunnel/freifunk-berlin-issue767-56c9a9d-tplink_archer-c7-v4-sysupgrade.bin

SvenRoederer, 2020-05-05:
@ikreb7  any news?

Mask66, 2020-05-07:
Same thing happens to me on sysupgrade (and the enforced assistant run) on an TP-Link TL-WDR4900 v1 with build Freifunk Berlin Dev-daily-1907 7542d4b.

SvenRoederer, 2020-05-09:
> Tried Builds 63,64, couldn't get to the webinterface to test, I just get permission denied when trying to access luci.

This should be fixed in recent master-branches (issue #788)

SvenRoederer, 2020-05-09:
> Same thing happens to me on sysupgrade (and the enforced assistant run) on an TP-Link TL-WDR4900 v1 with build Freifunk Berlin Dev-daily-1907 [7542d4b](https://github.com/freifunk-berlin/firmware/commit/7542d4b4518d83d60c09c07fa42e0d561587c8e1). (Build 67)
> 
> Edit: doesn't happen on Freifunk Berlin Dev-Monthly-Master [e099564](https://github.com/freifunk-berlin/firmware/commit/e0995641516e965e29c634041a09118d8001302b) (Build 62)

* Dev-Monthly-Master (e099564): has external repos at 30.4.2020
* Dev-daily-1907 (7542d4b): has external repos at 6.5.2020

So it's not a huge list of changes ....

```
config.mk
-OPENWRT_COMMIT=5e8b50da156f9b8fe1b17fa62a446c0a07734e76
+OPENWRT_COMMIT=5feb0df9bb3bd83ec49af17d2abc261fc58e2551

feeds.conf
-packages ^cfde14a8cf439518350c0de7308c498177c2cce4
+packages ^72e4e36a35ff081fb1c9e6404bf5afee6698d44a
-luci ^d320179573c8befc51559a9adc5a5820c561af13
+luci ^39a82906a1b92c8bb90e884e5a6fd41b420a9344
-routing ^3f8571194c2765ed31aa73459e86c2ebf943d27d
+routing ^9b42e24a54f03ebb6f58224b49036e8f739b175f
-packages_berlin ^c3c0f4b86b2f0812f8c3737ada35ee3cb02fe1fd
+packages_berlin ^a96afe33df8a371dc7b9657b874db92a022c03a9
```

Both devices are dual-band, right?

Mask66, 2020-05-09:
> Both devices are dual-band, right?
right, had an inkling that it could have something to do with the 802.11s configuration. 5Ghz radio was non-functional after skipping the assistant, but I didn't try real hard to get it to work.

SvenRoederer, 2020-05-09:
There is a known issue with some ath10k chips (#696), but the WDR4900 v1 is having older hardware (ath9k, I assume) .

ilippert, 2020-06-11:
Hardware `gl-mt300n-v2 router` with

reproduce

1. install (git-19.249.30590-1c7b1d6) / Freifunk Berlin Dev-daily-1907 99d94ca
2. configure freifunk, reboot
3. enter freifunk assistant in order to change bandwith.
4. click onto "save & reboot" 

produces

```
...b/lua/luci/model/cbi/freifunk/assistent/wireless.lua:143: attempt to concatenate field 'mode' (a nil value)
stack traceback:
	...b/lua/luci/model/cbi/freifunk/assistent/wireless.lua:143: in function 'i'
	/usr/lib/lua/luci/model/uci.lua:203: in function 'foreach'
	...b/lua/luci/model/cbi/freifunk/assistent/wireless.lua:102: in function 'write'
	...b/lua/luci/model/cbi/freifunk/assistent/wireless.lua:86: in function 'parse'
	/usr/lib/lua/luci/cbi.lua:201: in function 'parse'
	/usr/lib/lua/luci/cbi.lua:201: in function 'parse'
	/usr/lib/lua/luci/cbi.lua:559: in function 'parse'
	/usr/lib/lua/luci/dispatcher.lua:1216: in function '_form'
	/usr/lib/lua/luci/dispatcher.lua:848: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:412: in function </usr/lib/lua/luci/dispatcher.lua:411>
```

ikreb7, 2020-07-01:
Hello,

sorry for the late respond. I tested it with ``freifunk-berlin-dev-daily-1907-cb8a1a4-tplink_archer-c7-v4-sysupgrade`` and got the same error message:

````
...b/lua/luci/model/cbi/freifunk/assistent/wireless.lua:143: attempt to concatenate field 'mode' (a nil value)
stack traceback:
	...b/lua/luci/model/cbi/freifunk/assistent/wireless.lua:143: in function 'i'
	/usr/lib/lua/luci/model/uci.lua:203: in function 'foreach'
	...b/lua/luci/model/cbi/freifunk/assistent/wireless.lua:102: in function 'write'
	...b/lua/luci/model/cbi/freifunk/assistent/wireless.lua:86: in function 'parse'
	/usr/lib/lua/luci/cbi.lua:201: in function 'parse'
	/usr/lib/lua/luci/cbi.lua:201: in function 'parse'
	/usr/lib/lua/luci/cbi.lua:559: in function 'parse'
	/usr/lib/lua/luci/dispatcher.lua:1216: in function '_form'
	/usr/lib/lua/luci/dispatcher.lua:848: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:412: in function </usr/lib/lua/luci/dispatcher.lua:411>
```

Akira25, 2020-10-31:
@ikreb7 Could you please try one of these images? They are beta-images based on OpenWrt-19.07.4 I wonder, if that bug happens with them also.

http://download-master.berlin.freifunk.net/falter/

SvenRoederer, 2020-11-01:
@ikreb7 did you run a test with the build special debug-build mentioned in https://github.com/freifunk-berlin/firmware/issues/767#issuecomment-603150331?