freifunk-berlin/firmware #769: TP-Link CPE210/220: high CPU load: rssileds

everloop2, 2020-03-25:
**What is the problem?**
rssileds cause average of ~10% CPU load

**What is the expected behaviour?**
should be less to almost zero %

**Firmware Version:**
should be all  supporting CPE210/220, tested master (12.03.2020):

`CPU load:	CPU:   6% usr  21% sys   0% nic  70% idle   0% io   0% irq   1% sirq`

**Workaround 1**
disable rssileds -> frei.funk -> administration -> system -> startup -> stop rssileds / disable (at boot) -> idle gets around 95% :)

**Workaround 2**
remove package rssileds -> /firmware/configs -> ath79-generic.config -> CONFIG_PACKAGE_rssileds=n

CPU load been addressed in past, see:
https://github.com/openwrt/openwrt/pull/169
https://github.com/freifunk-gluon/gluon/commit/9c0a0898ea08dd8d88da805e9ae9212029cb730c

**IMHO**
-> signal strength leds are nice - average of 10% CPU not,
-> think moste usecases adjusting antenna for fine tuning people log in to box and use values shown in LuCI
-> would disable it or tune led signal refresh interval

booo, 2020-03-25:
This should be addressed upstream e.g. in the OpenWRT default configuration. 

Could you test the current behaviour with image from the OpenWRT development branch?

everloop2, 2020-03-25:
OpenWrt SNAPSHOT, r12706-1fb3c003d6 (25.03.2020) -> CPU load seems okay
```
Mem: 22204K used, 36240K free, 104K shrd, 2120K buff, 6560K cached
CPU:   0% usr   1% sys   0% nic  98% idle   0% io   0% irq   0% sirq
Load average: 0.02 0.04 0.04 1/38 1625
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
 1392     1 root     S     1440   2%   0% /usr/sbin/rssileds wlan0 200000 1 tp-link:green:link1 1 100 0 1 tp-link:green:link2 30 100
 1494     1 dnsmasq  S     1376   2%   0% /usr/sbin/dnsmasq -C /var/etc/dnsmasq.conf.cfg01411c -k -x /var/run/dnsmasq/dnsmasq.cfg014
   96     2 root     IW       0   0%   0% [kworker/0:1-eve]
```

everloop2, 2020-03-25:
Openwrt 19.07.02 (29.02.2020), r10947-65030d81f3 -> CPU load seems okay -> tested wifi uplink
```
Mem: 23720K used, 35832K free, 364K shrd, 2364K buff, 7844K cached
CPU:   0% usr   0% sys   0% nic  99% idle   0% io   0% irq   0% sirq
Load average: 0.18 0.22 0.14 1/41 2822
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
  124     2 root     IW       0   0%   0% [kworker/0:1]
 1432     1 root     S     1444   2%   0% /usr/sbin/rssileds wlan0 200000 1 tp-link:green:link1 1 100 0 1 tp-link:green:link2 30 100
 1573  1562 root     R     1212   2%   0% top
```

SvenRoederer, 2020-03-25:
Can you compare (diff) the led-settings in /etc/config/system between vanilla OpenWrt and the config causing 10% load?

everloop2, 2020-03-26:
setting:
```
config rssid 'rssid_wlan0'                
        option dev 'wlan0' -> all rssi LED are OFF (only LAN and power ON)
```
to:
```
config rssid 'rssid_wlan0'                
        option dev 'wlan0-mesh-2' or option dev 'wlan0-dhcp-2' -> rssi LED are ON (all LED are ON)
```
-> CPU load gets "normal" (box needs reboot)

-> quirk: if set ```option dev 'wlan0-dhcp-2'``` and wlan0-dhcp-2 (AP) is disabled in LuCI,
```
Mem: 33076K used, 26588K free, 972K shrd, 3992K buff, 11520K cached
CPU:   6% usr  22% sys   0% nic  69% idle   0% io   0% irq   0% sirq
Load average: 0.84 0.57 0.26 1/55 4453
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
 2218     1 root     S     1672   3%   1% /usr/sbin/olsrd -f /var/etc/olsrd6.conf -nofork
 2049     1 root     S     2012   3%   1% /usr/sbin/olsrd -f /var/etc/olsrd.conf -nofork
  124     2 root     IW       0   0%   1% [kworker/0:1]
 3080  2716 root     R     1212   2%   0% top
    8     2 root     IW       0   0%   0% [kworker/u2:1]
 2715  1021 root     S     1148   2%   0% /usr/sbin/dropbear -F -P /var/run/dropbear.1.pid -p 22 -K 300 -T 3
    7     2 root     SW       0   0%   0% [ksoftirqd/0]
 2321     1 root     SN    4304   7%   0% /usr/sbin/collectd -C /tmp/collectd.conf -f
 1808     1 root     S     2276   4%   0% /usr/sbin/uhttpd -f -h /www -r cottbus-lausi36 -x /cgi-bin -u /ubus -t 60 -T 30 -k 20 -A 1
  899     1 root     S     2088   3%   0% /sbin/rpcd -s /var/run/ubus.sock -t 30
 1079     1 root     S     1748   3%   0% /sbin/netifd
    1     0 root     S     1564   3%   0% /sbin/procd
 1116     1 root     S     1448   2%   0% /usr/sbin/odhcpd
 2531     1 root     S     1444   2%   0% /usr/sbin/rssileds wlan0-dhcp-2 200000 1 tp-link:green:link1 1 100 0 1 tp-link:green:link2
 1775     1 dnsmasq  S     1364   2%   0% /usr/sbin/dnsmasq -C /var/etc/dnsmasq.conf.cfg01411c -k -x /var/run/dnsmasq/dnsmasq.cfg014
 2641     1 root     S     1284   2%   0% /usr/sbin/pingcheck
  874     1 root     S     1248   2%   0% /sbin/logd -S 64
  525     1 root     S     1232   2%   0% /sbin/ubusd
 2716  2715 root     S     1220   2%   0% -ash
```
-> ```option dev 'wlan0-mesh-2'``` -> and wlan0-mesh-2 is disabled -> same behavior.

if wlan0-dhcp-2 or wlan0-mesh-2 is enabled again -> reproducible:

```
Mem: 33436K used, 26228K free, 976K shrd, 3992K buff, 11524K cached
CPU:   0% usr   4% sys   0% nic  93% idle   0% io   0% irq   1% sirq
Load average: 0.56 0.54 0.31 1/56 6241
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
  899     1 root     S     2088   3%   1% /sbin/rpcd -s /var/run/ubus.sock -t 30
 1808     1 root     S     2296   4%   1% /usr/sbin/uhttpd -f -h /www -r cottbus-lausi36 -x /cgi-bin -u /ubus -t 60 -T 30 -k 20 -A 1
 2049     1 root     S     2012   3%   1% /usr/sbin/olsrd -f /var/etc/olsrd.conf -nofork
 6124     1 root     S     1780   3%   1% /usr/sbin/hostapd -s -P /var/run/wifi-phy0.pid -B /var/run/hostapd-phy0.conf
 2321     1 root     SN    4304   7%   1% /usr/sbin/collectd -C /tmp/collectd.conf -f
  124     2 root     IW       0   0%   1% [kworker/0:1]
 2218     1 root     S     1672   3%   0% /usr/sbin/olsrd -f /var/etc/olsrd6.conf -nofork
 5790  2716 root     R     1212   2%   0% top
 2715  1021 root     S     1148   2%   0% /usr/sbin/dropbear -F -P /var/run/dropbear.1.pid -p 22 -K 300 -T 3
 2531     1 root     S     1444   2%   0% /usr/sbin/rssileds wlan0-dhcp-2 200000 1 tp-link:green:link1 1 100 0 1 tp-link:green:link2
  525     1 root     S     1232   2%   0% /sbin/ubusd
  543     1 root     S     1028   2%   0% /sbin/urngd
    7     2 root     SW       0   0%   0% [ksoftirqd/0]
    8     2 root     IW       0   0%   0% [kworker/u2:1]
 1079     1 root     S     1748   3%   0% /sbin/netifd
    1     0 root     S     1564   3%   0% /sbin/procd
 1116     1 root     S     1448   2%   0% /usr/sbin/odhcpd
 1775     1 dnsmasq  S     1364   2%   0% /usr/sbin/dnsmasq -C /var/etc/dnsmasq.conf.cfg01411c -k -x /var/run/dnsmasq/dnsmasq.cfg014
 2641     1 root     S     1284   2%   0% /usr/sbin/pingcheck
  874     1 root     S     1248   2%   0% /sbin/logd -S 64
 2716  2715 root     S     1220   2%   0% -ash
```

stock Openwrt 19.07.02 and master rssi sections are identical:

```
config rssid 'rssid_wlan0'
        option dev 'wlan0'
        option refresh '200000'
        option threshold '1'

config led 'led_lan'
        option name 'LAN'
        option sysfs 'tp-link:green:lan'
        option trigger 'netdev'
        option mode 'link tx rx'
        option dev 'eth0'

config led 'led_rssilow'
        option sysfs 'tp-link:green:link1'
        option trigger 'rssi'
        option iface 'rssid_wlan0'
        option minq '1'
        option maxq '100'
        option offset '0'
        option factor '1'

config led 'led_rssimediumlow'    
        option sysfs 'tp-link:green:link2'
        option trigger 'rssi'  
        option iface 'rssid_wlan0'
        option minq '30'                  
        option maxq '100'      
        option offset '0'               
        option factor '1'                 
                                
config led 'led_rssimediumhigh'   
        option sysfs 'tp-link:green:link3'
        option trigger 'rssi'
        option iface 'rssid_wlan0'        
        option minq '60'                  
        option maxq '100'         
        option offset '0'         
        option factor '1'                 
                             
config led 'led_rssihigh'         
        option sysfs 'tp-link:green:link4'
        option trigger 'rssi' 
        option iface 'rssid_wlan0'        
        option minq '80'                  
        option maxq '100'         
        option offset '0'         
        option factor '1'
```

SvenRoederer, 2020-03-26:
think the relates to https://github.com/freifunk-berlin/firmware/issues/509

SvenRoederer, 2020-03-26:
> ```
> config rssid 'rssid_wlan0'                
>         option dev 'wlan0' -> all rssi LED are OFF (only LAN and power ON)
> 
> config rssid 'rssid_wlan0'                
>         option dev 'wlan0-mesh-2' or option dev 'wlan0-dhcp-2' -> rssi LED are ON (all LED are ON)
> ```

To make it short: having a non-existing interface specified in "config rssid" will cause high CPU-load and as per #509 not working LEDs. Right?

everloop2, 2020-03-27:
> To make it short: having a non-existing interface specified in "config rssid" will cause high CPU-load and as per #509 not working LEDs. Right?

Exactly.

strange is -> vanilla OpenWrt -> `option dev 'wlan0'` is set -> all LED are on + normal CPU load

suggest:
-> `option dev 'wlan0-mesh-2'` -> if there is mesh node around rssi led will show it
-> AP signal strengh can be determined by clients connected or via LuCI, if AP is down clients will "show"/cant connect -> doesnt matter if rssi LED are on or off

Akira25, 2021-04-25:
This bug got fixed by https://github.com/Freifunk-Spalter/packages/pull/169 Maybe you want to adopt this solution.