freifunk-berlin/firmware #770: TP-Link CPE210/220: freifunk-berlin-development-572abd8: build failure

everloop2, 2020-03-25:
**What is the problem?**

build data_file_clashes:

```
Collected errors:
 * check_data_file_clashes: Package kmod-ath10k-ct-smallbuffers wants to install file /home/human/Freifunk/firmware_master/firmware/openwrt/build_dir/target-mips_24kc_musl/root-ath79/lib/modules/4.14.171/ath10k_core.ko
        But that file is already provided by package  * kmod-ath10k-ct
 * check_data_file_clashes: Package kmod-ath10k-ct-smallbuffers wants to install file /home/human/Freifunk/firmware_master/firmware/openwrt/build_dir/target-mips_24kc_musl/root-ath79/lib/modules/4.14.171/ath10k_pci.ko
        But that file is already provided by package  * kmod-ath10k-ct
 * opkg_install_cmd: Cannot install package kmod-ath10k-ct-smallbuffers.
```

**Firmware Version:**

master_freifunk-berlin-development-572abd8 (12.03.2020)

**Workaround**

/firmware/configs/ath79-generic.config -> CONFIG_PACKAGE_kmod-ath10k-ct-smallbuffers=n
-> firmware building :)

-> https://openwrt.org/packages/pkgdata/kmod-ath10k-ct-smallbuffers
ath10k-ct driver optimized for CT ath10k firmware (small buffers for low-RAM devices)

**Workaround2** (not tested)

/firmware/configs/ath79-tiny.config	-> CONFIG_PACKAGE_kmod-ath10k-ct=n
-> not tested due no profiles till ath79 migration

SvenRoederer, 2020-03-25:
I've seen this too. Fro me running "cd openwrt; make clean" fixed it.
running "make clean" should also do the same.

everloop2, 2020-03-25:
K, good to know
- sometimes do a make in /openwrt to get build files to add/test targets not available in freifunk-berlin/firmware

-> wizard	and mesh working -> master seems in good condition :)

closed 2020-03-25 by everloop2

SvenRoederer, 2020-03-25:
>     * sometimes do a make in /openwrt to get build files to add/test targets not available in freifunk-berlin/firmware

have a look at 04c3b4f96d7a0d1ceeeaa79f08c3b9e061bbe5b7 - this makes the imagebuilder capable of creating a firmware for any board supported by OpenWrt without having to compile yourself.

> 
> -> wizard and mesh working -> master seems in good condition :)

Thanks for the test