freifunk-berlin/firmware #771: TP-Link CPE210/220: berlin-development-572abd8: not sharing internet connection

everloop2, 2020-03-25:
**What is the problem?**

-> ffuplink is enabled and shows connected

and
```
Thu Mar 12 16:56:08 2020 daemon.err collectd[2522]: ping_sendto: Permission denied
Thu Mar 12 16:56:08 2020 daemon.err collectd[2522]: ping plugin: ping_send failed: Permission denied
```

**What is the expected behaviour?**

-> ffuplink should be disabled? at boot

-> ping should be disabled in mesh mode?

**Firmware Version:**
freifunk-berlin-development-572abd8

SvenRoederer, 2020-03-25:
Is this a dual or single-LAN-port board? 
The PoE-port will be freifunk-LAN by default.

Probably relates to: https://github.com/freifunk-berlin/firmware/issues/292; https://github.com/freifunk-berlin/firmware-packages/pull/185; https://github.com/freifunk-berlin/firmware-packages/pull/190; https://github.com/freifunk-berlin/firmware/issues/465#issuecomment-309100088

everloop2, 2020-03-26:
https://openwrt.org/toh/tp-link/cpe210

its CPE210 v2, one LAN Port

SvenRoederer, 2020-05-09:
So this LAN-port is Freifunk-DHCP - net by default and an uplink should only be found via mesh.
Did you change the meaning of the LAN-port to be WAN?

everloop2, 2020-05-09:
> So this LAN-port is Freifunk-DHCP - net by default
yes, works like expected

> uplink should only be found via mesh
yes, FreifunkWizard -> choosed: not sharing internet connection

> Did you change the meaning of the LAN-port to be WAN?
no, tested mesh only mode -> if WAN is needed it has to be set up after FFwizard via wifi due one LAN port

its a major in master branch -> FreifunkWizard -> choosed: not sharing internet connection:

-> FFuplink is started at boot
-> WAN is started at boot
-> pingcheck tries to ping WAN

SvenRoederer, 2020-05-09:
> edit: WAN is not there if "not sharing internet connection" (like expected)

Upstream-OpenWrt at least defines a LAN-network (which we use as Freifunk-DHCP later) and if there are enough physical ports they will also define a WAN-network. So for single-port devices there is currently no WAN at all. 
That's a thing that can be solved with some work on out configtools-package.

pmelange, 2020-05-10:
> its a major in master branch -> FreifunkWizard -> choosed: not sharing internet connection:
> 
> -> FFuplink is started at boot

PR https://github.com/freifunk-berlin/firmware-packages/pull/190 would have fixed this.

everloop2, 2020-05-12:
https://github.com/freifunk-berlin/firmware-packages/pull/185
https://github.com/freifunk-berlin/firmware-packages/pull/190

what about using both solutions? -> "sharing internet" disabled by default

"sharing internet" only activated by ffwizard choice

and/or installing uplink package
- if uninstalling uplink package "sharing internet" should be disabled again

SvenRoederer, 2020-05-12:
As far as I remember, each PR has its own drawback. We should decide, which one is causing less problems (now and also for the code maintenance at all).

pmelange, 2020-05-13:
freifunk-berlin/firmware-packages#185 fails the test case described here https://github.com/freifunk-berlin/firmware-packages/pull/185#issuecomment-474150669 and is in my option not an option.

freifunk-berlin/firmware-packages#190 was commented on and the comment answered here https://github.com/freifunk-berlin/firmware-packages/pull/190#discussion_r361880433