freifunk-berlin/firmware #772: Build process fails: nchname.txt.txt: No such file or directory

daftmilk, 2020-03-27:
Hello!

I am trying to build version 1.05 to update from Freifunk Berlin Hedy 1.0.4 v1.0.4 on Mac Os 10.15.3

The build fails with:

```
cd XXXXX/tmp/gluon_.stamp-gluon-module-openwrt; git name-rev --refs openwrt-* $(grep _COMMIT XXXXX/.stamp-gluon-module-openwrt | cut -d "=" -f 2) | cut -d / -f 2 | cut -d \~ -f 1 nchname.txt.txt
cut: nchname.txt.txt: No such file or directory
make: *** [.stamp-gluon-module-openwrt] Error 1
```

It looks like it would be pretty easy to fix, but I just don't know where to look.

SvenRoederer, 2020-03-27:
this relates to 95dd252b34e79057a284bc895d85809752cd4147; #738; #739; #740 

> 
> Unfortunately there is no precompiled image for my router:
> http://buildbot.berlin.freifunk.net/buildbot/stable/1.0.5/ar71xx-generic/
> 

Have you tried using the imagebuilder, as per https://github.com/freifunk-berlin/firmware/blob/master/README.md#customizing-make.
Sadly the precompiled imagebuilder has been deleted on download-server, but you can still find it at http://77.245.36.54:8000/firmwares/stable/ar71xx-generic/

SvenRoederer, 2020-03-27:
> The build fails with:
>
> ```
> cd XXXXX/tmp/gluon_.stamp-gluon-module-openwrt; git name-rev --refs openwrt-* $(grep _COMMIT XXXXX/.stamp-gluon-module-openwrt | cut -d "=" -f 2) | cut -d / -f 2 | cut -d [...]

On first view it looks like a MacOS-specific Escaping-issue of the master branch. Seems you did not checkout the "v1.0.5" tag.

SvenRoederer, 2020-04-10:
> On first view it looks like a MacOS-specific Escaping-issue of the master branch.

PR #781 will fix this also.

@daftmilk did you check out the correct branch before you started the build?

SvenRoederer, 2020-05-08:
Should be fixed with merging PR https://github.com/freifunk-berlin/firmware/pull/781. 
Feel free to reopen if you still encounter this problem.

closed 2020-05-08 by SvenRoederer