freifunk-berlin/firmware #773: Feature: Turn on/off (performance consuming) services with a switch

znrR, 2020-03-29:
As an advanced and pro user I want to turn off/on (performance consuming) services such as DHCP, IPV6 and others [needs further spec] per Interface with one switch, without the need to switch on/off that services at various spots.

Solution Idea: Realised via scripts that turn off that services at all required spots and an overview of services in menu services to "Interfaces" in "Networks". To identify which services consume a lot of performance a hint in that overview would be great.

SvenRoederer, 2020-03-29:
It there a specific kind of performance-gap you found / referencing to? IMHO it will be hard to find a universal setup of services an advanced user don't want in his individual setup.

everloop2, 2020-03-31:
@znrR see: https://github.com/freifunk-berlin/firmware/issues/769#issuecomment-603955561

if everything runs "normal" and box has nothing to do CPU load is around 0-5% for single core boxes running vanilla OpenWrt:

https://openwrt.org/toh/astoria/arv752dpw22#hardware_highlights -> EasyBox803 -> 333MHz

https://openwrt.org/toh/tp-link/cpe210#hardware_highlights -> TP-Link CPE210/CPE220 -> 650 MHz

Freifunk adds some services - more then 10% CPU spikes are unusual if box idles

see: http://monitor.berlin.freifunk.net/ -> system loads

or some testing - EasyBox803 doing mesh, wifi RaLink RT3062F is not full supported and EasyBox hardware "a bit crappy"

http://monitor.berlin.freifunk.net/detail.php?p=load&t=load&h=cottbus-lausi36-2&s=86400

if its about slow web interface LuCI -> try faster development builds ;)

znrR, 2020-04-07:
OK - that's really interesting! I have a few examples to provide (although i don't know how to collect the data) where a nanostation m2 loco or an TP-Link TL-WR1043N/ND v1 get so slow (by luci?!) that even tcp/ip traffic slows down dramatically, sometimes even crash. Perry helped me with the m2 loco and set a scheduled task to` */5 * * * *	rm -f /tmp/olsrd*core` which helped a bit. lets talk about it on tuesday, to see if the feature request is something different like "make luci less performace consuming" :) the original feature request came from mathias @kls0e  who also has experience with slow downed routers. can you add some of your examples as well?

kls0e, 2020-04-07:
why yes, for instance, there are situations where I want routers to only mesh, to not provide ap functionality. I am using this for some older 1043 nodes, they do serve well as mesh nodes.  
That means a feature request consists of the option of disabling the ap interface by switch (maybe mesh as well?). What is the best way to do so? uci command? For the time being, i'd conf it in /etc/config/network as `option disabled '1'`.

Also a common practice are 24h reboots of routers by a cronjob, let's add a switch for that as well. 
Here is a suitable crontab line, will reboot the router at 4AM and will touch banner for system time reasons:

`0 4 * * * sleep 70 && touch /etc/banner && reboot >/dev/null 2>&1`

in terms of general stability I prefer to disable uhttpd as described [here](https://openwrt.org/docs/guide-user/luci/luci.secure#more_secure_configuration):

```
/etc/init.d/uhttpd disable
/etc/init.d/uhttpd stop

If you need access to LuCI webserver, just start uhttpd service without enabling it from SSH and access it via SSH-tunnel.

/etc/init.d/uhttpd start

When you are done with LuCI webserver, don't forget to stop uhttpd service.

/etc/init.d/uhttpd stop
```
Not sure this a one-way switch makes sense, if so, i'd call it a _TOUGHSwitch_. I'd use it tho.

SvenRoederer, 2020-04-07:
> nanostation m2 loco or an TP-Link TL-WR1043N/ND v1 get so slow (by luci?!)

You are aware, that those devices are 32MB RAM only? There is the 4/32-warning of OpenWrt and discouraging their general support and use. Please see also https://github.com/freifunk-berlin/firmware/pull/666. @kls0e your special configurations are probably also a candidate for a "tiny" imagetype.

znrR, 2020-04-07:
a tiny image type would be a great solution! I put it in for discussion for later today!