freifunk-berlin/firmware #775: dnsmasq: stop building with TFTP-support

SvenRoederer, 2020-04-05:
currently we use the package "dnsmasq" in our firmware. This version shows the following startup-message:

>Sun Apr  5 16:57:46 2020 daemon.info dnsmasq[18954]: started, version 2.80 cachesize 150
Sun Apr  5 16:57:46 2020 daemon.info dnsmasq[18954]: compile time options: IPv6 GNU-getopt no-DBus no-i18n no-IDN DHCP no-DHCPv6 no-Lua TFTP no-conntrack no-ipset no-auth no-DNSSEC no-ID loop-detect inotify dumpfile

Probably it's worth to disable TFTP-support to save some flash-mem. This is probably most relevant for tiny devices (save some flash, but also RAM)

pmelange, 2020-04-07:
I disagree with this.  I personally have used TFTP from a freifunk router multiple times and find it very useful.

SvenRoederer, 2020-06-26:
I just checked, removing TFTP-support saves 3,5 kb of package-size. 
But there is no direct way to disable tftp-support in dhsmasq package. such option is only provided for dnsmasq-full. But it will become to confusing to include dnsmasq-full on "tiny" devices, as it has less features than dnsmasq.

closed 2020-06-26 by SvenRoederer