freifunk-berlin/firmware #776: Disabling Smart Gateway by default for BB only usecase (or fix it)

znrR, 2020-04-07:
I kept receiving strange routes in my setup (not sharing internet) and learned that smart gateway was the problem. Now I disable Smart Gateway on all routers when they are only disconnected to BB and not sharing internet. Maybe it  makes sense to deactivate it by default for "am FF Netz teilnehmen" setups.

znrR, 2020-04-07:
maybe relates to this one here: https://github.com/freifunk-berlin/firmware/issues/549

SvenRoederer, 2020-04-07:
@znrR  what means "strange routes"? Isn't it a principle of Freifunks Mesh-routing, that every node can contribute to the "global" routing-table. In consequence this is also valid for bad routing-rules based on a faulty setup of a node.

* SmartGateway: Isn't is that you can choose between problems caused by the "hackish" Smartgateway-solution or problems as of not using the SmartGateyway-feature (metric based routing).
* SGw respect Bandwidth: as far as I heard, SGW is using ETX for selecting the Gateway. It that your intention to change?
* Drop of OLSRd: I made several PRs to remove some dependencies to OLSRd (see "olsr"-tag of our issues / PRs, https://github.com/freifunk/openwrt-packages). There is a consensus to drop olsr ...

znrR, 2020-04-07:
@pmelange implements a new page that appears right after the installation wizard and is integrated under the freifunk menu "Erweiterte Einstellungen" which allows to turn Smart Gateway on/off properly and will also be the spot to place other options such as "BBdigger on/off". @znR will provide translations, @znrR @xa-y-ax will test

znrR, 2020-04-21:
State of this feature on 21. April:
- UI proposal to switch tunnel and related options is welcomed
- currently its "just" HTML UI
- logic behind needs to be developed
[image]
[image]
[image]

znrR, 2020-04-21:
next steps:
- develop technical concept on how to write / apply these settings @pmelange 
- write better and comprehensive texts in english and then translate in multiple languages: @znrR @xa-y-ax