freifunk-berlin/firmware #780: buildenv: using an forked openwrt repo and branch from github fails

pmelange, 2020-04-09:
I would like to use a forked repo and a test branch to compile a test version of openwrt.  Unfortunately it does not work.

Here are the errors
```
rm -f /home/pmelange/src/mt7621/openwrt/.config                                
rm -f .stamp-
rm -f .stamp-gluon-module-openwrt
echo >>.stamp-gluon-module-openwrt "OPENWRT_REPO=https://github.com/pmelange/openwrt.git"
echo >>.stamp-gluon-module-openwrt "OPENWRT_COMMIT=641703b11b7b4e0e17298bbe5386ae302d9ed955"
git clone $(grep _REPO .stamp-gluon-module-openwrt | cut -d "=" -f 2) /home/pmelange/src/mt7621/tmp/gluon_.stamp-gluon-module-openwrt                          
Cloning into '/home/pmelange/src/mt7621/tmp/gluon_.stamp-gluon-module-openwrt'...
remote: Enumerating objects: 32, done.
remote: Counting objects: 100% (32/32), done.
remote: Compressing objects: 100% (24/24), done.
remote: Total 504820 (delta 11), reused 16 (delta 8), pack-reused 504788
Receiving objects: 100% (504820/504820), 175.36 MiB | 4.92 MiB/s, done.        
Resolving deltas: 100% (344264/344264), done.                                  
cd /home/pmelange/src/mt7621/tmp/gluon_.stamp-gluon-module-openwrt; git name-rev --refs openwrt-* $(grep _COMMIT /home/pmelange/src/mt7621/.stamp-gluon-module-openwrt | cut -d "=" -f 2) | cut -d / -f 2 | cut -d \~ -f 1 >branchname.txt     
cd /home/pmelange/src/mt7621/tmp/gluon_.stamp-gluon-module-openwrt; grep -q undefined branchname.txt  || \
        printf >>/home/pmelange/src/mt7621/.stamp-gluon-module-openwrt "OPENWRT_BRANCH=%s\n" \
                $(echo  | tr '[:lower:]' '[:upper:]') \                        
                $(cat branchname.txt)
rm -rf /home/pmelange/src/mt7621/tmp/gluon_.stamp-gluon-module-openwrt         
rm -f /home/pmelange/src/mt7621/modules
cat >>/home/pmelange/src/mt7621/modules .stamp-gluon-module-openwrt            
cat >>/home/pmelange/src/mt7621/modules .stamp-gluon-module-packages .stamp-gluon-module-luci .stamp-gluon-module-freifunk .stamp-gluon-module-routing .stamp-gluon-module-packages_berlin .stamp-gluon-module-gluon                           
echo >>/home/pmelange/src/mt7621/modules GLUON_FEEDS=\' packages luci freifunk routing packages_berlin gluon  \'
--- Updating module 'openwrt' ---
Initialized empty Git repository in /home/pmelange/src/mt7621/openwrt/.git/    
remote: Enumerating objects: 10, done.
remote: Counting objects: 100% (10/10), done.
remote: Compressing objects: 100% (10/10), done.
remote: Total 487662 (delta 2), reused 2 (delta 0), pack-reused 487652
Receiving objects: 100% (487662/487662), 171.07 MiB | 5.01 MiB/s, done.        
Resolving deltas: 100% (331972/331972), done.                                  
From https://github.com/pmelange/openwrt                                       
 * branch            HEAD       -> FETCH_HEAD                                  
fatal: Not a valid branch point: '641703b11b7b4e0e17298bbe5386ae302d9ed955'.   
unable to find commit "641703b11b7b4e0e17298bbe5386ae302d9ed955" on branch "" in repo "https://github.com/pmelange/openwrt.git".                               
make: *** [Makefile:136: .stamp-pre-patch] Error 1                             
```

If I use a commit ID which is on the master branch of my forked repo, it works, but if the commit is in a branch, I get the errors above.

I originally tried using https://github.com/vDorst/openwrt/commit/70188cf0837e7991a8eea390fbd75d89062941b3 and it failed with the same kind of error as above.  So I added vDorst as a new remote, fetched the changes, cherry-picked into a new branch and tried from my forked repo.  

I would really like a fix to this problem so that I can continue to develop and test.

@SvenRoederer you have made a lot of changes to the build environment.  Can you please help?

pmelange, 2020-04-09:
I have merged the commits in the branch to my master, just so that I can continue with testing.  It's working for now, but it shouldn't be

pmelange, 2020-04-09:
If you want to try this at home,

git checkout daily/upstream-master

in config.mk
```
...
OPENWRT_SRC=https://github.com/vDorst/openwrt.git
OPENWRT_COMMIT=70188cf0837e7991a8eea390fbd75d89062941b3
...
```

make TARGET=ramips-mt7621

sarumpaet, 2020-04-09:
If you set `OPENWRT_BRANCH=ramips-5.4` in config.mk as well, it works. There's some magic in our Makefile that looks like it should be doing that, but obviously it's broken (it surely looks _very_ brittle anyways - I'd put such stuff outside of a Makefile as formatting and escaping _will_ bite you).

pmelange, 2020-04-09:
It looks like it should.... :/

$ cat config.mk 
```
# default parameters for Makefile
SHELL:=$(shell which bash)
TARGET=ath79-generic
PACKAGES_LIST_DEFAULT=notunnel tunnel-berlin-tunneldigger manual
OPENWRT_SRC=https://github.com/vDorst/openwrt.git
OPENWRT_BRANCH=ramips-5.4
OPENWRT_COMMIT=70188cf0837e7991a8eea390fbd75d89062941b3
SET_BUILDBOT=env
MAKE_ARGS=
```

```
remote: Counting objects: 100% (2/2), done.
remote: Compressing objects: 100% (2/2), done.
remote: Total 496077 (delta 0), reused 0 (delta 0), pack-reused 496075
Receiving objects: 100% (496077/496077), 173.01 MiB | 2.54 MiB/s, done.        
Resolving deltas: 100% (337983/337983), done.                                  
cd /home/pmelange/src/mt7621/tmp/gluon_.stamp-gluon-module-openwrt; git name-rev --refs openwrt-* $(grep _COMMIT /home/pmelange/src/mt7621/.stamp-gluon-module-openwrt | cut -d "=" -f 2) | cut -d / -f 2 | cut -d \~ -f 1 >branchname.txt     
cd /home/pmelange/src/mt7621/tmp/gluon_.stamp-gluon-module-openwrt; grep -q undefined branchname.txt  || \
        printf >>/home/pmelange/src/mt7621/.stamp-gluon-module-openwrt "OPENWRT_BRANCH=%s\n" \
                $(echo  | tr '[:lower:]' '[:upper:]') \                        
                $(cat branchname.txt)
rm -rf /home/pmelange/src/mt7621/tmp/gluon_.stamp-gluon-module-openwrt         
rm -f /home/pmelange/src/mt7621/modules
cat >>/home/pmelange/src/mt7621/modules .stamp-gluon-module-openwrt            
cat >>/home/pmelange/src/mt7621/modules .stamp-gluon-module-packages .stamp-gluon-module-luci .stamp-gluon-module-freifunk .stamp-gluon-module-routing .stamp-gluon-module-packages_berlin .stamp-gluon-module-gluon                           
echo >>/home/pmelange/src/mt7621/modules GLUON_FEEDS=\' packages luci freifunk routing packages_berlin gluon  \'
--- Updating module 'openwrt' ---
Reinitialized existing Git repository in /home/pmelange/src/mt7621/openwrt/.git/
From https://github.com/vDorst/openwrt
 * branch                  HEAD       -> FETCH_HEAD                            
fatal: Not a valid branch point: '70188cf0837e7991a8eea390fbd75d89062941b3'.   
unable to find commit "70188cf0837e7991a8eea390fbd75d89062941b3" on branch "" in repo "https://github.com/vDorst/openwrt.git".                                 
make: *** [Makefile:136: .stamp-pre-patch] Error 1                             
```

sarumpaet, 2020-04-09:
Weird - that fixed it for me. You aren't using a Mac by chance? Perhaps add some debug echo lines to `scripts/update.sh` to do tracing.

pmelange, 2020-04-09:
No, I'm using linux on the machine where I build everything else.  I'll take a look at it sometime, but not today.

pmelange, 2020-04-09:
The branch "ramips-5.4" has been updated.  I tried it again with @sarumpaet's suggestion, and it didn't work. :(  Same error as last time.

```
--- Updating module 'openwrt' ---
Initialized empty Git repository in /home/pmelange/src/mt7621/openwrt/.git/    
remote: Enumerating objects: 2, done.
remote: Counting objects: 100% (2/2), done.
remote: Compressing objects: 100% (2/2), done.
remote: Total 487092 (delta 0), reused 0 (delta 0), pack-reused 487090
Receiving objects: 100% (487092/487092), 170.58 MiB | 6.61 MiB/s, done.        
Resolving deltas: 100% (331623/331623), done.                                  
From https://github.com/vDorst/openwrt
 * branch            HEAD       -> FETCH_HEAD                                  
fatal: Not a valid branch point: '9e59838fc68b75d2849ea7ba81de8869c71cd2df'.   
unable to find commit "9e59838fc68b75d2849ea7ba81de8869c71cd2df" on branch "" in repo "https://github.com/vDorst/openwrt.git".                                 
make: *** [Makefile:136: .stamp-pre-patch] Error 1                             

```

SvenRoederer, 2020-04-09:
@pmelange  https://github.com/freifunk-berlin/firmware/blob/b71cbcc7f1cc0f8395c8a18f86a286c6f4cf0251/Makefile#L84 should be the source of the problem you are reporting. It's trying to find a branchname where this commits belong to and stores this in the ```branchname.txt``` file. This code has been primarily tested on official OpenWrt-branches only.

As mentioned by @sarumpaet the code added in 7a72a84291e9944f187548e7de35406c3df57f8d (PR #714) to generate the ```modules``` file (needed by the gluon-build) from our current ```config.mk``` and ```feeds.conf``` is nothing to be proud of.

SvenRoederer, 2020-04-09:
I had a look at the referenced code of the repo of @vDorst.
These commits can also just copied (git format-patch) into ```patches/openwrt/``` and will be applied onto the upstream-repo defined in config.mk.

SvenRoederer, 2020-04-10:
PR #781 will fix this problem by avoiding the "very brittle" code.

pmelange, 2020-04-10:
I'll give PR #781 a try.

SvenRoederer, 2020-05-08:
Should be fixed with merging PR https://github.com/freifunk-berlin/firmware/pull/781. 
Feel free to reopen if you still cant use other repos.

closed 2020-05-08 by SvenRoederer