freifunk-berlin/firmware #783: uplink/notunnel: use label-mac as unique part of ffuplinks-dhcpname

SvenRoederer, 2020-04-10:
in https://github.com/freifunk-berlin/firmware-packages/commit/8e2b55e1e4e9b396562cb0305cdecd912a5f7cda we use the random MAC-address to have a 2nd unique name when registering in DHCP-server.
in OpenWrt a function to retrieve the MAC-address printed on the outside label was added (https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=0340718863040919ecaba74b2173ed52ebc194ce - after OpenWrt-19.07 branch of). 
Let's use this address in the uplinks hostname. 

So the host can easily looked up in the upstream-LAN, esp. when more than one Router is installed.

pmelange, 2020-04-12:
This is an interesting idea.  Buy why not just go with something human readable like a concatenation of the hostname and "ffuplink"?

I assume you mean setting the value for network.ffuplink.hostname

Akira25, 2020-04-13:
I'd also prefer a solution which is human-readable without further need to lookup something. Unless there is a good point, why the ffuplink-hostname needs to be machine readable.

SvenRoederer, 2020-04-13:
The point is: when the defaults-script is running the hostname is still "gib-mir-einen-namen". This will also not be unique and also will cause name-clashes.

SvenRoederer, 2020-04-13:
Btw: You are aware, what we use at the moment? The currently used random string can't even be lookup easily, beside checking the config of all connected routers.

pmelange, 2020-04-13:
Well, I'm still of the opinion that ffuplink should be disabled until after the router runs through the wizard. See https://github.com/freifunk-berlin/firmware/issues/603