freifunk-berlin/firmware #784: buildenv: all possible target images are built in openwrt/bin/targets/ARCH/SUBARCH

pmelange, 2020-04-12:
The current buildenv creates a sysupgrade and factory image for every possible router in openwrt/bin/targets/ARCH/SUBARCH.

I beilieve this is an artiface of using the following options in common.config

CONFIG_TARGET_MULTI_PROFILE=y
CONFIG_TARGET_ALL_PROFILES=y

The images being created seem to be the default image sets from openwrt.  Here is the manifest...
[freifunk-berlin-next-875e6ff-ramips-mt7621.manifest.txt](https://github.com/freifunk-berlin/firmware/files/4467160/freifunk-berlin-next-875e6ff-ramips-mt7621.manifest.txt)

pmelange, 2020-04-12:
The full list of generated images can be seen here
[sha256sums.txt](https://github.com/freifunk-berlin/firmware/files/4467170/sha256sums.txt)

SvenRoederer, 2020-04-13:
> The images being created seem to be the default image sets from openwrt. Here is the manifest...
> [freifunk-berlin-next-875e6ff-ramips-mt7621.manifest.txt](https://github.com/freifunk-berlin/firmware/files/4467160/freifunk-berlin-next-875e6ff-ramips-mt7621.manifest.txt)

The packgelist is based on the packages we set to "y" in configs/common.config.

SvenRoederer, 2020-11-01:
> The images being created seem to be the default image sets from openwrt. Here is the manifest...

No, it's a specialized packagelist (based on all packages enabled with option "=y" in the config) which results in an initrd-image with a basic UI to flash the final sysupgrade. Esp. usefull for ERX.

---

But what's the intention of this issue?

closed 2020-11-02 by pmelange