freifunk-berlin/firmware #787: Nodes with unconfigured name keep the template "gib-mir-einen-namen"

SvenRoederer, 2020-04-18:
### Bug report

**What is the problem?**

When the default name of an unconfigured node is not changed in the wizard it stays untouch after finishing the setup.

[image]

**What is the expected behaviour?**
There is code in the wizard which should use a MESH-ipaddress in this case. This seems to be broken.

**Firmware Version:**
Hedy 1.0.2-rc1 / 52c60b6 (I cant tell if there are local modifications, as I collected this data from hopglass only)

Akira25, 2020-04-19:
As far as I can remember, it worked for me recently on a hedy-1.0.4 device. I will check that for you.

Akira25, 2020-04-19:
okay, I flashed a GL-AR150 with hedy-1.0.4 (tunneldigger) and gave the wizard only the coordinates, the speed of the internet connection and the okay for the monitor.
Afterwards things worked as i expected: Map, monitor and everything else was fine.

The node does not rename it self after its mesh-address, but after the gateway-address of its DHCP-Network.

[image]
[image]
[image]
[image]

SvenRoederer, 2020-04-20:
Was also working on a recent "master" build. So unsure, how these node made it onto the map. Probably the node was not configured by the wizard?
I wrote a PM to the nodes contact address to check ... let's see.

Akira25, 2020-04-21:
I think we should close this issue, as the rest of the problem seems not to be firmware-related (as proven by to poeple).

closed 2020-04-21 by SvenRoederer