freifunk-berlin/firmware #789: buildenv: new "modules" file based feed syntax does not allow for alternative feed types

pmelange, 2020-04-26:
In the old system, using the "feeds.conf" file, it was possible to use feed types other than "src-git".  For example, what I had used frequently to do prototyping, was the "src-link" feed type.  All of the supported feed types which openwrt supports are listed here: https://openwrt.org/docs/guide-developer/feeds

This issue was created by request from @SvenRoederer (https://github.com/freifunk-berlin/firmware/pull/781#issuecomment-619441557)

SvenRoederer, 2020-04-26:
> In the old system

This refers to "before e81f4109c431d70fa79908df99f49704f14a9bbc" was merged to implement PR #714.

---

Before the change we had our feeds.conf directly linked to the OpenWrt-root. Since the change the ```scripts/update.sh``` and ```scripts/feeds.sh``` generating the feeds.conf in OpenWrt-root. 
Where do you see the failure? During the update.sh or the feeds.sh?
Can you provide a log?

SvenRoederer, 2020-04-26:
* Not sure if this is supported by gluon at all. (These scripts have be provided by gluon)
* The referenced repos in  modules-file will be checked out locally in the packages folder, before referencing them in OpenWrt-root. I wonder if the ```git remote add file:/``` syntax will work?