freifunk-berlin/firmware #791: Enhancement: Use the freifunk favicon.png for the luci interface

pmelange, 2020-04-29:
I recommend using the freifunk favicon for the web interface. 

[image]

To do this, I would recommend putting this into the $OPENWRT_DIR/files/www/luci-static/bootstrap folder to overwrite the original.

[image]

Original size: 733 bytes
Freifunk site:1528 bytes (possibly this can be optimized).

Akira25, 2020-04-29:
I like that idea. Lets do it.

pmelange, 2020-04-29:
PR submitted https://github.com/freifunk-berlin/firmware/pull/792

SvenRoederer, 2021-07-18:
Done by switching to theme "Freifunk-generic" in 02cfecc51c47d82062f350bef96959a605888821

closed 2021-07-18 by SvenRoederer