freifunk-berlin/firmware #798: dispatcher.lua errors

ilippert, 2020-05-12:
this is my first report here - i hope I followed to reporting protocol sufficiently well.

### Bug report

**What is the problem?**
flashing new image does not work. It has not been working for a while. But I do not recall when it started.

**What is the expected behaviour?**
Using `/cgi-bin/luci/admin/system/flashops` to flash the current image should result in  a clean reboot into the new firmware.

Instead, I get this:

´´´
/usr/lib/lua/luci/dispatcher.lua:247: attempt to call field 'striptags' (a nil value)
stack traceback:
	/usr/lib/lua/luci/dispatcher.lua:247: in function 'd'
	/usr/lib/lua/luci/dispatcher.lua:585: in function 'menu_json'
	/usr/lib/lua/luci/dispatcher.lua:671: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:412: in function </usr/lib/lua/luci/dispatcher.lua:411>
´´´

This occurs on a `gl-mt300n-v2` router, flashing from

`Freifunk Berlin Dev-daily-1907 ef318a5` to

http://buildbot.berlin.freifunk.net/buildbot/unstable/ramips-mt76x8/85/notunnel/freifunk-berlin-dev-daily-1907-1c92c67-gl-mt300n-v2-sysupgrade.bin, without keeping the (not anyhow customised) configuration

**Firmware Version:**
http://buildbot.berlin.freifunk.net/buildbot/unstable/ramips-mt76x8/85/notunnel/freifunk-berlin-dev-daily-1907-1c92c67-gl-mt300n-v2-sysupgrade.bin

**Site Configuration:**
the router is not configured at all.

SvenRoederer, 2020-05-13:
> attempt to call field 'striptags' (a nil value)

that's caused by "https://github.com/freifunk-berlin/firmware/blob/ef31a4c83e4c27003ab5a9a6af180ba9a74a473f/patches/packages/luci/0202-split-luci-base_PR2817.patch" (backport of https://github.com/openwrt/luci/pull/2817)

SvenRoederer, 2020-05-13:
just seen that your current image ef318a5 is very old: "committed on 27 Dec 2019". Can you flash a current one via console and retry?

SvenRoederer, 2020-05-13:
Found a missed "util.striptags()" which should be fixed in 8ad23ec046979ea570f4a4321b646ba9a16ecdf0.

Please install fresh build from branch "daily/upstream-1907" and test again.

ilippert, 2020-06-11:
Sorry, still getting  
```
...b/lua/luci/model/cbi/freifunk/assistent/wireless.lua:143: attempt to concatenate field 'mode' (a nil value)
stack traceback:
	...b/lua/luci/model/cbi/freifunk/assistent/wireless.lua:143: in function 'i'
	/usr/lib/lua/luci/model/uci.lua:203: in function 'foreach'
	...b/lua/luci/model/cbi/freifunk/assistent/wireless.lua:102: in function 'write'
	...b/lua/luci/model/cbi/freifunk/assistent/wireless.lua:86: in function 'parse'
	/usr/lib/lua/luci/cbi.lua:201: in function 'parse'
	/usr/lib/lua/luci/cbi.lua:201: in function 'parse'
	/usr/lib/lua/luci/cbi.lua:559: in function 'parse'
	/usr/lib/lua/luci/dispatcher.lua:1216: in function '_form'
	/usr/lib/lua/luci/dispatcher.lua:848: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:412: in function </usr/lib/lua/luci/dispatcher.lua:411>
```
after tying "applying" the changes of how much bandwith I change, using the freifunkassistent

> Found a missed "util.striptags()" which should be fixed in [8ad23ec](https://github.com/freifunk-berlin/firmware/commit/8ad23ec046979ea570f4a4321b646ba9a16ecdf0).
> 
> Please install fresh build from branch "daily/upstream-1907" and test again.

ilippert, 2020-06-11:
thanks a lot

ilippert, 2020-06-11:
Sorry, no, updating the config and "save & reboot" still gets me

```
...b/lua/luci/model/cbi/freifunk/assistent/wireless.lua:143: attempt to concatenate field 'mode' (a nil value)
stack traceback:
	...b/lua/luci/model/cbi/freifunk/assistent/wireless.lua:143: in function 'i'
	/usr/lib/lua/luci/model/uci.lua:203: in function 'foreach'
	...b/lua/luci/model/cbi/freifunk/assistent/wireless.lua:102: in function 'write'
	...b/lua/luci/model/cbi/freifunk/assistent/wireless.lua:86: in function 'parse'
	/usr/lib/lua/luci/cbi.lua:201: in function 'parse'
	/usr/lib/lua/luci/cbi.lua:201: in function 'parse'
	/usr/lib/lua/luci/cbi.lua:559: in function 'parse'
	/usr/lib/lua/luci/dispatcher.lua:1216: in function '_form'
	/usr/lib/lua/luci/dispatcher.lua:848: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:412: in function </usr/lib/lua/luci/dispatcher.lua:411>
```

ilippert, 2020-06-11:
And when I click elsewhere, and then try to apply changes, I get this response

`Failed to confirm apply within 30s, waiting for rollback…`

SvenRoederer, 2020-06-11:
```
> ...b/lua/luci/model/cbi/freifunk/assistent/wireless.lua:143: attempt to concatenate field 'mode' (a nil value)
> stack traceback:
```
Seems to be exact the same as reported in #767. can you please add there how you triggered this (steps to reproduce), what hardware and which Firmware-version you used?

As the initial issue "striptags" was solved, I'll close this here.

closed 2020-06-11 by SvenRoederer