freifunk-berlin/firmware #810: collectd-dnsmasq: compiling fails on latest openwrt master

PolynomialDivision, 2020-06-18:
I wanted to use the firmware-packages repro with current OpenWrt master.
This include does not work:

https://github.com/freifunk-berlin/firmware-packages/blob/master/addons/collectd-addons/files/collectd-dnsmasq/postinst.sh#L3

```
cp -fpR /home/nick/openwrt/build_dir/target-mips_24kc_musl/root-ath79 /home/nick/openwrt/build_dir/target-mips_24kc_musl/root.orig-ath79
/home/nick/openwrt/build_dir/target-mips_24kc_musl/root-ath79/tmp/collectd-dnsmasq-addon_postinst.sh: line 3: /lib/functions.sh: No such file or directory
postinst script ./usr/lib/opkg/info/collectd-dnsmasq-addon.postinst has failed with exit code 1
```

Could we maybe enable the issues in the firmware-packages repro? I think that would be better. :)

SvenRoederer, 2020-06-18:
> Could we maybe enable the issues in the firmware-packages repro? I think that would be better. :)

Can you open an separate issue for this? I also like the idea, but remember some others argumented against.

SvenRoederer, 2020-06-18:
I assume you see this problem with the imagebuilder, right?

PolynomialDivision, 2020-06-18:
> I assume you see this problem with the imagebuilder, right?

Yep. Sry, I should add information how to reproduce.

Adding:
```
src-git freifunk https://github.com/freifunk/openwrt-packages.git
src-git freifunkberlin https://github.com/freifunk-berlin/firmware-packages.git
```
into feeds.conf and select the packages, e.g. collectd-dnsmasq-addon.

Has the berlin freifunk image builder a special openwrt install part?

SvenRoederer, 2020-06-18:
Since some time the imagebuilder runs th postinst automatically (see relating issues: https://github.com/freifunk/openwrt-packages/issues/10, https://github.com/openwrt/luci/pull/1635#issuecomment-385257774) and in this script we seem to have an absolute path which will run on the build-host file-system.

PolynomialDivision, 2020-06-18:
> Since some time the imagebuilder runs th postinst automatically (see relating issues: [freifunk/openwrt-packages#10](https://github.com/freifunk/openwrt-packages/issues/10), [openwrt/luci#1635 (comment)](https://github.com/openwrt/luci/pull/1635#issuecomment-385257774)) and in this script we seem to have an absolute path which will run on the build-host file-system.

So actually we need to find a way to disable them on build process? Or just removing like u did with policy routing?

PolynomialDivision, 2020-06-18:
We could place some condition in the script like

```
[ "$ACTION" = ifup ] || exit 0
. /lib/functions.sh
```

https://github.com/openwrt/packages/blob/master/net/chrony/files/chrony.hotplug#L19-L22

SvenRoederer, 2020-06-18:
it depends on how they are written. 
When they call a uci-default script it can be dropped, when running an other specialized script it should probably only executed when running on real hardware.
For reference see https://github.com/freifunk-berlin/firmware-packages/pull/199 which deals a very similar issue and has a comment to the relating OpenWrt-commit.

PolynomialDivision, 2020-06-18:
Can we maybe just add some
```
if [ ! -f "/lib/functions.sh" ]; then
  exit 0
fi
```
as workaround, so the packages compile? Just tested the compiling and it worked! :)
For me it is important that I can use the packages to test them and I'm afraid by removing post install hookes, I could break the packages. :S

SvenRoederer, 2020-06-25:
As it's broken package (https://github.com/freifunk-berlin/firmware-packages/pull/205#issuecomment-649836032) I'm fine with keeping this issue open, as reminder or hint ...

closed 2020-10-13 by PolynomialDivision