freifunk-berlin/firmware #813: structure: packaging important patches

PolynomialDivision, 2020-06-19:
My aim is to use the official imagebuilder to build freifunk firmware. Can we maybe package the relevant patches and add them to https://github.com/freifunk-berlin/firmware-packages?

SvenRoederer, 2020-06-19:
With the patches in the firmware-repo we fix / modify code, which is not in our control. So how should a file from our feed modify the imagebuilder or the kernel ?

PolynomialDivision, 2020-06-20:
> With the patches in the firmware-repo we fix / modify code, which is not in our control. So how should a file from our feed modify the imagebuilder or the kernel ?

Are all that patches really relevant and necessary? Can we maybe put the relevant stuff upstream?

SvenRoederer, 2020-06-20:
There was recently a discussion on this (https://github.com/freifunk-berlin/firmware/issues/774).
Most of the patches are either:
* FreifunkBerlin specific changes of upstream-defaults
* backports from Upstream master to 19.07 (which will not be backported by upstream)
* changes where nobody hat the energy to discuss with upstream

PolynomialDivision, 2020-06-21:
The discussion is so long. Could u just provide me the changes you want to have upstream? :) I will take care of that, if I think they are useful.

>     * backports from Upstream master to 19.07 (which will not be backported by upstream)

Since I want to do OpenWrt master builds, I do not care. ;) And the next release will have all changes included again.

closed 2020-10-13 by PolynomialDivision